opendatacam/opendatacam #377: Performance Problems with Jetson Nano

studwieska4344, 2021-03-05:
Hello,

I repeatedly have performance problems using the Jetson Nano. 

The demo video runs, but I notice huge fluctuations in the FPS. The values differ from 1-2 PFS, 4-5 FPS to 11 FPS, while the setup remains unchanged.

I wasn´t able to find the reason for these fluctuations so far.

Is there already a similar problem known or can you think of a solution to my problem?

Thank you in advance!

vsaw, 2021-03-05:
There have been FPS drops reported with Jetpack 4.4. Which version are you running? 

See also https://github.com/opendatacam/opendatacam/issues/233

studwieska4344, 2021-03-12:
Hi,
updating the jetpacks did indeed help. 
Thanks for the help!

closed 2021-03-12 by studwieska4344