opendatacam/opendatacam #379: Overhead detection of bikes and persons

jvdhurk, 2021-03-11:
With the camera in an overhead position OpenDataCam (or the model currently used) performs very well when it comes to counting cars. But it has a hard time detecting persons and bikes. Less problems when persons and (especially) bikes are viewed from the side.

Is there a better model or a model that is specifically trained on vehicles from all different angles?

vsaw, 2021-03-12:
What do you mean by overhead?

If the camera is turned upside down, you should be able to flip it in gstreamer using https://gstreamer.freedesktop.org/documentation/videofilter/videoflip.html?gi-language=c. (I believe there is even a Nvidia version of this to take advantage of HW acceleration)

jvdhurk, 2021-03-12:
"overhead" means that the camera is above the area to be observed and the camera is facing downwards.

It is very clear that the current model can recognize the side view of a bike very well, but the front view and the top view is a challenge.

tdurand, 2021-03-13:
We noticed that also.. OpenDataCam is based on the default weights file of YOLOv4 , which is trained on the MS-COCO dataset.. the only way to improve the detection is to train your own YOLO weights file including more example images of bikes and persons in other scenario, which is not straightforward to do .. you need lots of data + experience in training large models..  I have sucessfully retrained YOLO to detect some other objects, but something very simple that didn't need huge dataset.

For YOLOv3 there was some weights file trained on OpenImage (another dataset), but it is not available for YOLOv4 , that said I don't know if it was better for your use cases

Also didn't dedicate the time yet but gonna update OpenDataCam to use the new Scaled YOLOv4 architecture:  which improve a little bit the accuracy of the neural network:.. but as it is still trained on the same dataset (MS-COCO), I don't expect it will improve much your scenario.

You can learn and ask more around training and accuracy directly in the darknet repository: https://github.com/alexeyab/darknet

closed 2021-03-13 by tdurand