opendatacam/opendatacam #381: Give bounding box color of predicted class

tychotdeback, 2021-03-28:
Being able to give the dashed bounding box line a specific color based on the predicted class would be nice to have. Say you have a vehicle type which rarely occurs. By being able to give the bounding box line a red color for that specific class, this would be immediately visually intuitive when looking at the live view. Could this be implemented in "LiveViewEngine.js"?

vsaw, 2021-03-29:
It should be possible. I believe you'd have to change some of those lines here: https://github.com/opendatacam/opendatacam/blob/development/components/canvas/engines/LiveViewEngine.js#L32-L47.

For a quick fix you could hard code the colors in the file and see if that works :)

tychotdeback, 2021-03-30:
Thanks a lot for replying! Much appreciated. I will try to hard code the colors in the indicated segment and see how it goes :-)