opendatacam/opendatacam #383: Jetson Xavier NX (JetPack 4.5.1) compatibility issue

zeljkomarinkovic, 2021-04-08:
For few days now, I'm trying to solve this problem. I know that in docs says that we should use JetPack 4.3 but there is no version for NX, first one is 4.4. 

If I try to install it with docker, ODC works fine but something during installation breaks terminal and become unusable with message 

`There was an error creating the child process for this terminal. Failed to open PTY: Permission denied.`

if I'm running it without docker all is perfect until `npm start` where it hangs on `Success init db`

Mongodb works fine and added url in config.json. 
Made sure that have corect path to darknet, added yolov4-416x416.cfg to it also tested on video, stream and image, works fine

goodfellasthai, 2021-04-08:
Try this I see a similar thing on my Jetbot Nano when using Jupyter and regularly have to issue the below command.

sudo chmod 666 /dev/pts/ptmx

vsaw, 2021-04-08:
This is odd. I have a fork of ODC running on JetPack 4.5.1 just fine. I wonder if this is related to ODC or caused by the underlying OS. Have you asked around in the Jetson Forums?

zeljkomarinkovic, 2021-04-09:
I have no experience with docker at all so I'm maybe doing something wrong but again, I'm following documentation here and every time same story. But running without docker works though

> Have you asked around in the Jetson Forums?

I have opened [this thread on Jetson Forum](https://forums.developer.nvidia.com/t/system-update-breaks-terminal/170860)

vsaw, 2021-04-09:
@zeljkomarinkovic 

This post hints at a  Kernel driver issue: https://forums.developer.nvidia.com/t/system-update-breaks-terminal/170860/8?u=vsaw
Are you running a fresh install of JetPack or have you done any modifications prior to installing ODC?

> I have no experience with docker at all so I'm maybe doing something wrong but again

Please keep in mind that ODC can not access the CSI-Camera (e.g. Raspi-Cam) from within Docker. USB-Cams or Files work fine though.

zeljkomarinkovic, 2021-04-09:
yep, on fresh install of JetPack, no prior modification to installing ODC

tdurand, 2021-04-11:
I saw you solved the problem ? (you edited the first issue) .. The solution was to start OpenDataCam by pinging localhost:8080 ? 

feel free to re-open otherwise

closed 2021-04-11 by tdurand