opendatacam/opendatacam #387: Bike project with Jetson Nano

willytrueba, 2021-04-21:
Hello again,

Thank you first for answering and your time. I'm still working on the bike counting project, but I run into several problems. At the moment I use a Jetson Nano with USBCam, and after installing it, everything works ok a couple of times. If I turn off and on a couple of times I have the following error and I don't know how to solve it.

```
Start recording
already started
First request on console stream
Something went wrong: connect ECONNREFUSED 172.18.0.3:8070
Too many retries, YOLO took more than 3 min to start, likely an error
0
Something went wrong: connect ECONNREFUSED 172.18.0.3:8070
Too many retries, YOLO took more than 3 min to start, likely an error
0
New client, close previous stream
New client, close previous stream
already started
Something went wrong: connect ECONNREFUSED 172.18.0.3:8070
Too many retries, YOLO took more than 3 min to start, likely an error
0
```

Also the Ubuntu terminal does not work either ...

My configuration is this:

```
NVIDIA Jetson Nano (Developer Kit Version)
   * Jetpack 4.3 [L4T 32.3.1]
   * NV Power Mode: 5W - Type: 1
   * jetson_stats.service: active
 - Libraries:
   * CUDA: 10.0.326
   * cuDNN: 7.6.3.28
   * TensorRT: 6.0.1.10
   * Visionworks: 1.6.0.500n
   * OpenCV: 4.1.1 compiled CUDA: NO
   * VPI: 0.1.0
   * Vulkan: 1.1.70
```

I am working so that a daemon every x minutes uploads the data to a server and clears the logs, so that ODC is not loaded with much information.

You recommend that I change Jetson, or use another tool. And is ODC going to have a version soon?

Greetings and thank you

tdurand, 2021-04-22:
Hi, 

Changing jetson won't help 😉. Turning opendatacam on and off isn't properly tested so it might crash at some point or not shutdown the process properly....

I can't really help more without beeing able to reproduce (you would need to tell us the exact steps and procedure leading to that bug).. I guess the best way would be to associate shutting down jetson with turning off opendatacam... so you do you make sure everything is cleared up...

tdurand, 2021-04-22:
also I see you jetson is on 5W power mode.. maybe you want to try to set it to 10W to see if that happens also..

tdurand, 2021-04-22:
ODC is gonna have a new release sometime before this summer, no exact date to share.

vsaw, 2021-04-22:
@willytrueba Acking what @tdurand just said, this will not be solved by a bigger Jetson. My guess is there are two things here that we need to look at

1. Power Mode: 5W Power Mode lowers CPU freq etc. so it could be that Darknet is just taking to long to start in 5W mode. Try going to 10W mode following these instructions: https://www.jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/ 
2. The message essentially means that the Darknet process  could not be started. Besides power this could be caused by a flacky video device. Can you confirm that the USB-Camera is plugged in and working outside of ODC?

willytrueba, 2021-04-22:
Hi @vsaw and @tdurand !! Thanks for your answers.

USB camera (Logitech C270) is working correctly, maybe should be change the resolution? As for changing the mode from 5W to 10W, I'm going to try it.

These days I am going to test the Jetson Nano in real mode. It is installed to count vehicles and bikes on a road, in remote mode. For this I have used Balena to install a simple ODC that can be configured remotely. I have used an external MongoDB Atlas server for the data and camera will start recording at 8:00 until 20:00. The rest of the day will be on stand-by but without turning off.

After 6 hours the temperature (62º) and the memory (45%) are correct. But I have two problems/issues

- Due to the location of the camera, at a height of 4 meters pointing to a road, the bicycles are counted as pedestrians, I don't know if it could be configured.

- On the other hand I find a data delay between 10 and 15 minutes from Jetson Nano to the MongoDB Atlas server. This is normal? And also that all the Tracker information is being sent and I don't need it. Can I configure to not send tracker data?

I wait for your comments :)

Best

vsaw, 2021-04-22:
> Hi @vsaw and @tdurand !! Thanks for your answers.
> 
> USB camera (Logitech C270) is working correctly, maybe should be change the resolution? As for changing the mode from 5W to 10W, I'm going to try it.

Would you mind posting the complete log you see during boot? Then we can check why Darknet is not starting.

> These days I am going to test the Jetson Nano in real mode. It is installed to count vehicles and bikes on a road, in remote mode. For this I have used Balena to install a simple ODC that can be [...]

I'm actually working on a simple backend service for ODC to help retrieve logs from devices. If you are interested ping me up at valentin@nono.cam

> After 6 hours the temperature (62º) and the memory (45%) are correct. But I have two problems/issues

Should be no problem. The temperature is well within the working range of Jetson Nano (~80°C)

> * Due to the location of the camera, at a height of 4 meters pointing to a road, the bicycles are counted as pedestrians, I don't know if it could be configured.

This is due to the Model performance. You would have to train a new/better model or change the camera position for better results.

> * On the other hand I find a data delay between 10 and 15 minutes from Jetson Nano to the MongoDB Atlas server. This is normal? And also that all the Tracker information is being sent and I don't need it. Can I configure to not send tracker data?

Unfortunately the tracker collection can not be turned off. However the development branch of ODC has a few improvements to the database which reduces the load on the database.

willytrueba, 2021-04-22:
> Would you mind posting the complete log you see during boot? Then we can check why Darknet is not starting.

Now, I cannot have the complete log, because I have installed the Balena Image for Jetson Nano, and I don't have any error when start. But I will buy another jetson nano to test without Balena Image.
> 
> > These days I am going to test the Jetson Nano in real mode. It is installed to count vehicles and bikes on a road, in remote mode. For this I have used Balena to install a simple ODC that can be configured remotely. I have used an external MongoDB Atlas server for the data and camera will start recording at 8:00 until 20:00. The rest of the day will be on stand-by but without turning off.
> 
> I'm actually working on a simple backend service for ODC to help retrieve logs from devices. If you are interested ping me up at [valentin@nono.cam](mailto:valentin@nono.cam)

I will write to you because I'm very interested.
> Should be no problem. The temperature is well within the working range of Jetson Nano (~80°C)
>
>
> This is due to the Model performance. You would have to train a new/better model or change the camera [...]

I don't know how to train a new/better model, but I think that the camara position "see" a narrow and tall rectangle, and think that is a person, not a bike. Do you know if exists a manual to learn how is the better configuration/position for a camara?
> Unfortunately the tracker collection can not be turned off. However the development branch of ODC has a few improvements to the database which reduces the load on the database.

I think that could be an option for a new version disable save the tracker information on mongodb because for 300 detected objects, I have 200,000 trackers records, and if you have an external database is a lot of information sending ...

tdurand, 2021-04-22:
> I don't know how to train a new/better model, but I think that the camara position "see" a narrow and tall rectangle, and think that is a person, not a bike. Do you know if exists a manual to learn how is the better configuration/position for a camara?

only way is to retrain the model.. which is not an easy task to do but can be learned, search for "Train custom dataset YOLOv4" on google.. this video sounds nice: https://www.youtube.com/watch?v=N-GS8cmDPog 

> I think that could be an option for a new version disable save the tracker information on mongodb because for 300 detected objects, I have 200,000 trackers records, and if you have an external database is a lot of information sending ...

you can disable it my modifying the code.. but yes will add to the list of features to add for next version..

willytrueba, 2021-04-22:
> only way is to retrain the model.. which is not an easy task to do but can be learned, search for "Train custom dataset YOLOv4" on google.. this video sounds nice: [...]

Ohh ... I have to learn a lot of to this :)

> you can disable it my modifying the code.. but yes will add to the list of features to add for next version..

Can you send me more information where can modify the code? I need to compile or only change the code?

willytrueba, 2021-04-23:
I try to modify the server/db/DBManager.js, but I don't have any idea how its working. I change the file, restart ODC and all my chnages are losing ... How can I modify its? I'm working with Balena image.

vsaw, 2021-04-23:
@willytrueba If I remember correctly Balena uses the Docker Version. So you would have to

1. modify the code
2. build a new docker image
3. deploy on device

willytrueba, 2021-04-23:
Ok, if I understar, download with git the opendatacam project, modify, upload and build with balena, and after install on device?

tdurand, 2021-04-24:
Unfortunately it is very time consuming to explain the codebase... I would say you need to understand what you are doing and it is not a simple copy paste and follow step job and we can't really assist you on every step.... that said we are open to do some paid consulting if you have a budget for your project. we also also open to answer on github specific questions on bugs or struggle you can have but this need to be something we can answer in 5 min.

To give you a pointer here is the line you should uncomment on the development branch to disable persisting tracker data : https://github.com/opendatacam/opendatacam/blob/development/server/db/MongoDbManager.js#L308 , didn't test but should work

Same thing to generate a new balena image from the development branch is not documented step by step, you have also some pointers here: https://github.com/opendatacam/opendatacam/blob/master/documentation/jetson/CREATE_DOCKER_IMAGE_JETSON.md but this is not a all a guide you can follow blindly and everything will work.

Generally speaking to understand how OpenDataCam works and tweak it yourself you need to have good knowlege of how client server apps works, node.js and javascript + docker and have also played with darknet, you have a architecture schema here: https://github.com/opendatacam/opendatacam/blob/master/documentation/DEVELOPMENT_NOTES.md#technical-architecture

In the next 2  months (before summer) I'm gonna dedicate some time to OpenDataCam to do a v4 release and will integrate an option to disable to  tracker data in the config.json

have a great week-end

willytrueba, 2021-04-24:
Hi Thibault,

First of all thank you for your answers and time. I am very impressed with your work, and at no time I want to abuse your knowledge and time. My idea is to do everything myself, and in fact I will keep learning and testing, especially the git and docker processes for create Balena image.

The project I am developing is quite simple, as it is about counting bicycles for a few hours a day. I've been running it for four days in real time, and the conclusions I'm getting and would like to share with you:

- It's a 4GBs Jetson Nano, configured with Balena. The power is 5W, and I have installed a fan.

- The Jetson Nano works over 40% CPU and a temperature of 62º. Hard disk space is the same and is fine.

- The counting function works perfectly, although it confuses bicycles with pedestrians, although I think it is because of the location of the camera (I have to try to get a good position/angle) and detects all objects, even things like "backpack" or "fire hydrant".

- During these four days it has counted about 800 objects each day, but on the other hand the trackers have shot up to 800,000 records. This is a problem if you work with a database, because it performs hundreds of updates every second and slows down the update, becoming saturated. I have modified the application and no longer save the trackers in the database. The database performance is perfect now. And I am working on a module to store in an external mysql database that I will share with you.

- On the other hand, the memory of the Jetson is increasing little by little and after about four hours it saturates, causing it to reboot. My question is if the tracker information is in memory (I imagine that to make the heatmap) and if there is a way to clean it or just do not save it (it is an idea).

- Another action I want to perform are a couple of processes to start and shutdown the application at certain times. The startup process that starts ODC, reads the areas from a json file, creates them in ODC and starts recording, and another process that stops recording, cleans up all the information and stops ODC. As soon as I have those two processes I will pass them to you.

- One idea that comes to my mind is to query the "status" to get the temperature/cpu/memory of the ODC, or is this a crazy idea?

Above all I want to emphasize thanks for your help and comments, and that at no time I want to take advantage of your work, and that you can count on my collaboration.

Thank you

tdurand, 2021-04-27:
> On the other hand, the memory of the Jetson is increasing little by little and after about four hours it saturates, causing it to reboot. My question is if the tracker information is in memory (I [...]

No the tracker information isn't kept in memory... we just keep the past 300 frames, you can config that in the config.json

That said.. there may be still some memory leak but I don't think it comes from OpenDataCam node.js code.. can you monitor the details of the memory use and see which process cause memory increase ? Might be darknet we use to have problems with it.

In bash try to use some tool to monitor memory like `htop`

> One idea that comes to my mind is to query the "status" to get the temperature/cpu/memory of the ODC, or is this a crazy idea?

I guess you should query directly the jetson : https://github.com/rbonghi/jetson_stats .. 

> Above all I want to emphasize thanks for your help and comments, and that at no time I want to take advantage of your work, and that you can count on my collaboration.

Many thanks, appreciated, also credits goes to @vsaw that was lately more active than me on support 👏 . 

> The project I am developing is quite simple, as it is about counting bicycles for a few hours a day. I've been running it for four days in real time

We are super keen to here about the different usage of OpenDataCam , would be great if you can share a picture and detail a bit  you use case here: https://github.com/opendatacam/opendatacam/issues/84 if you feel like it. For v4 release want to showcase the diversity of projects that can benefit of OpenDataCam.

napking, 2021-05-10:
> I think that could be an option for a new version disable save the tracker information on mongodb because for 300 detected objects, I have 200,000 trackers records, and if you have an external database is a lot of information sending

I seem to have an alternate solution for this issue:
[](https://docs.mongodb.com/manual/reference/command/convertToCapped/)

Convert the mongoDB collection to a capped collection. This seems to retain documents only up to the set limit. Can be done from the mongoShell without modifying the code on your OpenDataCam device.

willytrueba, 2021-05-10:
> I seem to have an alternate solution for this issue:
> (https://docs.mongodb.com/manual/reference/command/convertToCapped/)
>
> Convert the mongoDB collection to a capped collection. This seems to retain [...]

Have 200,000 trackers records in MongoDB is not a problem, and your solution for "capped" its are really nice. My problem is with API calls to external MongoDB. I have a Jetson Nano connect to 4G Router, without a great connection, and made a lot of API calls by minute.

tdurand, 2021-05-22:
closing this and feature request to disable persisting every frame of the tracker is tracked 😄 here: https://github.com/opendatacam/opendatacam/issues/388

tdurand, 2021-05-22:
would love to have some more insight on your use case here @willytrueba ✌ : https://github.com/opendatacam/opendatacam/issues/84 we want to feature as much diversity of projects for v4 release

closed 2021-05-22 by tdurand