opendatacam/opendatacam #390: Add Ethics code of conduct / License

tdurand, 2021-04-27:
Pointers:

- https://www.licenses.ai/
- https://medium.com/ml5js/ml5-js-code-of-conduct-4eb8fcae1ef7

tdurand, 2021-05-10:
https://medium.com/ml5js/a-new-code-of-conduct-and-license-for-ml5-js-6b0e4c109b76

Aspie96, 2021-05-13:
Here is my response to licenses.ai.
I contacted them, but got no reply.

> Plese, don't refer to the license at https://www.licenses.ai/open-source-license as "Open Source".
> 
> The term Open Source has a specific meaning. The definition is at: https://opensource.org/osd and has a specific interpretation.
> 
> Such meaning is well-recognized: not only has the term "Open Source" being coined specifically to refer to it, the meaning is widely accepted by the community at large and even by governments, laws and some contracts.
> You can find more info here: https://opensource.org/authority
> 
> In addition, OSI is supported by all OSI Affiliates: https://opensource.org/affiliates which include many of the main organizations concearned with software and licensing such as, to name just a few:
> - Debian
> - Creative Commons
> - Eclipse
> - FreeBSD
> - Gnome
> - KDE e.V.
> - The Linux Foundation
> - Mozilla
> - Python Software Foundation
> - Software Freedom Conservancy
> - Perl foundation
> - Wikimedia Foundation
> - Wordpress Foundation
> 
> Such definition is also well understood by many developers.
> 
> 
> There is nothing wrong at all in using a non Open Source license.
> And there is absolutely nothing wrong in writing such a license.
> 
> But calling it "Open Source" is misleading, it is harmful to Open Source and it ican be harmful for what you are trying to create as well.
> Because confusion is never good for good ideas. Clarity is.
> 
> For this reason different non Open Source licensing schemes should have a different name.
> 
> Please see:
> - https://steveklabnik.com/writing/the-culture-war-at-the-heart-of-open-source
> "I think we’ll end up with a new movement. For the same reasons that “open source” came up with a new name, I think the movement that will arise from today’s developers will also need a new name."
> - https://anticapitalist.software/
> "The Anti-Capitalist Software License is not an open source software license."
> - https://coopcycle.org/en/software/
> "So the CoopCycle software is indeed not open source : its code is available on Github, but not anyone can run it for commercial purposes !"

Tensorflow is Open Source software and it respects the freedom of the user.
As a user, there is no way an ethical source license, which restricts what you do, will help you.
As a result, users will chose to use alternatives.

However, here is why you shouln't want such a license even if you don't have the benefit of the user (and them using the library) is not what you care about:
- For when you want your own government to behave: the government also makes laws, including the very law you rely upon for licenses. If it became a problem for the government, it would simply change the law. However, if you live in a Democracy, there is already a process to decide what the government should do! The power for that is not given to developers, it's given to the people.
- For when you want other governments to behave: other governments follow their own laws and oppressive ones do not care about your license. Even governments which do care still make laws.
- For when you want to prevent other individuals from doing that which is already illegal: it's already illegal. There is already a penalty for that. Now, having an ethical license might increase the penatly: now the license violator is guilty not only for what they do, but for copyright and contract violation. This makes no sense for two reasons. First, you, developer, will be the "victim" of the action, not the actual victim. Also, no matter how bad a crime is, randomly increasing the penalty is absurd. The law decides what's illegal and it also decides, already, what the right penalty for it is. Did you actually check the penalty for the crimes described in the license and told yourself: "yes! the penalty for this crime should be increased exactly by that of a copyright violation!". If so, advocate for increasing the penalty. But again, just because a crime is bad doesn't mean the penalty should be increased: it only means there is an ideal penalty and that such ideal penalty is very high. The current penalty could be below or above that.
- Let's say someone uses your library for something which the government partially pays for. Because the government already has an opinion about what is and what isn't allowed, to give public money to your project would not be very democratic and would make no sense from the prospective of the government, since you'll forbid usage of the project (thereby restricting unilaterally how the result of your work with that public money will be used). So ethical licenses cannot be used for publicly paid projects.
- Let's say you are making research. Research is about providing information to others. The role of researchers, why we have them in society, is not to tell others what to do, but to tell others truth. To see the universe and point where others haven't looked before. There is no reason researchers deserve to decide how results are used and if they do that's not truly progress.
- Even if you don't actually care about the drop in usage, you are not trying to restrict any government (not yours, not others), you are not forbidding anything already illegal, you are not getting governments funding and you are not doing research: congrats, that's a narrow option. But even then, even in that case, the best way to prevent things you don't like is trough legislation, rather than simply telling people they cannot do [... cut ...] h things with one particular tool.

Tensorflow was built for the Open Source community, by the Open Source community, with an Open Source license everybody can trust.
And it is used for reasearch, a place where Open Source is much needed, and less control would help.

Using a proprietary license would betray what Tensorflow is.

More info:
- https://perens.com/2019/09/23/sorry-ms-ehmke-the-hippocratic-license-cant-work/
- https://www.gnu.org/philosophy/programs-must-not-limit-freedom-to-run.html

Aspie96, 2021-05-13:
I find it interesting that you provided one of my tweets as a reference because I strongly believe such licenses would be a bad idea for Tensorflow.

I don't find them wrong when they don't try to pretend to change language and call themselves Open Source though

Aspie96, 2021-05-14:
[I deleted some comments. For some reason, upon finding this issue I thought I was in the Tensorflow.js repository. I got confused while having both repos open. I apologize].

I find it interesting you use my tweet as a reference, since I do not support this kind of licenses.

As for licenses.ai, they are not a reliable source because they misrepresent their licenses as Open Source while they do not actually qualify.
I contacted them about it and received no response. Here is my email to them:

> Plese, don't refer to the license at https://www.licenses.ai/open-source-license as "Open Source".
>
> The term Open Source has a specific meaning. The definition is at: https://opensource.org/osd and [...]

Now, this is not by itself a reason not to use an ethical license, nor is it the reason for my opposition.

I am not involved in this project and I have no problem if you decide to use an ethical license and not an Open Source license.
However, I respecfully suggest it would be better not to, here is why:

As a user, there is no way an ethical source license, which restricts what you do, will help you.
As a result, users will chose to use alternatives.

However, here is why you shouln't want such a license even if you don't have the benefit of the user (and them using the library) is not what you care about:

- For when you want your own government to behave: the government also makes laws, including the very law you rely upon for licenses. If it became a problem for the government, it would simply change the law. However, if you live in a Democracy, there is already a process to decide what the government should do! The power for that is not given to developers, it's given to the people.
- For when you want other governments to behave: other governments follow their own laws and oppressive ones do not care about your license. Even governments which do care still make laws.
- For when you want to prevent other individuals from doing that which is already illegal: it's already illegal. There is already a penalty for that. Now, having an ethical license might increase the penatly: now the license violator is guilty not only for what they do, but for copyright and contract violation. This makes no sense for two reasons. First, you, developer, will be the "victim" of the action, not the actual victim. Also, no matter how bad a crime is, randomly increasing the penalty is absurd. The law decides what's illegal and it also decides, already, what the right penalty for it is. Did you actually check the penalty for the crimes described in the license and told yourself: "yes! the penalty for this crime should be increased exactly by that of a copyright violation!". If so, advocate for increasing the penalty. But again, just because a crime is bad doesn't mean the penalty should be increased: it only means there is an ideal penalty and that such ideal penalty is very high. The current penalty could be below or above that.
- Let's say someone uses your library for something which the government partially pays for. Because the government already has an opinion about what is and what isn't allowed, to give public money to your project would not be very democratic and would make no sense from the prospective of the government, since you'll forbid usage of the project (thereby restricting unilaterally how the result of your work with that public money will be used). So ethical licenses cannot be used for publicly paid projects.
- Let's say you are making research. Research is about providing information to others. The role of researchers, why we have them in society, is not to tell others what to do, but to tell others truth. To see the universe and point where others haven't looked before. There is no reason researchers deserve to decide how results are used and if they do that's not truly progress.
- Even if you don't actually care about the drop in usage, you are not trying to restrict any government (not yours, not others), you are not forbidding anything already illegal, you are not getting governments funding and you are not doing research: congrats, that's a narrow option. But even then, even in that case, the best way to prevent things you don't like is trough legislation, rather than simply telling people they cannot do such things with one particular tool.

Respectfully.

vsaw, 2021-05-15:
@tdurand would you mind sharing your goals of what it is you want to achieve by relicensing OpenDataCam?

tdurand, 2021-05-16:
@vsaw , don't necessarly want to re-license OpenDataCam ... 

The general idea is to feature ethics statements for the projects, maybe just a code of conduct.. let's be honest it is mainly to raise awareness around privacy / human rights issues caused by massive automated surveillance "use case" like this: https://algorithmwatch.org/en/greek-camps-surveillance/ ,  or naive tech solutionism  that may do more harm than good.. _( for example I see a lot of mask detection on people and I think as a society we shouldn't need to have cameras starring at people to make sure they have mask on...)_

As a small OpenSource project we will never be able to enforce any of this.. and all the process that comes into deciding what is good or what is bad is super complex... it is a very blurry line..  but I think lots of people working on this field don't even consider whose aspects so if we can at least trigger some _"Hey, you know that computer vision tech is not neutral and maybe what you are working on could somehow harm some people in a way you didn't think of, now do what you want but we told you 😁"_

tdurand, 2021-05-17:
feel free to jump in also, made this issue public to make it an open discussion !

tdurand, 2021-05-17:
Just a quick heads up, thanks BTW @Aspie96 to have taken the time of jumping in... also after reviewing a bit the path the [ml5js project went](https://medium.com/ml5js/a-new-code-of-conduct-and-license-for-ml5-js-6b0e4c109b76) , which is an interesting read... maybe too "US centrict" lawyery way of doing things for us... 

After taking a few days to reflect and spoke with @b-g also, decided to just to include in a prominent way something like a "artist statement" on the website / documentation of OpenDataCam for next release... no re licensing, no code of conduct.. keep it very simple as always..

The content would be  reflect more of less what I wrote in the past comment, raise awareness about ethical issues caused by AI, and point to some learning ressources around those...

On that matter I saw that people behind the fast.ai library have done a whole course around ethics: https://ethics.fast.ai/ , which I didn't review yet, but I'm a big fan of their work.. so I'm pretty sure it is very good.

Aspie96, 2021-05-17:
First, let me say I am not familiar with this project and, in principle, the choice doesn't affect me.
I replied for two reasons: first, I made a mistake, and for some reason I thought this issue had been opened in a different project, then I updated the answer.
Secondly, one of my tweets was mentioned as a source.

One thing I'd like to point out is that my arguments only work in the case of licensing and only refer to licensing. They are completely meaningless in any other case.

In principle, no code of conduct at all hurts the Open Source nature of a project.
Indeed, there are projects with codes of conduct I strongly disagree with. But as a user it doesn't affect me at all, and all advantages of software freedom are still there.
Some code of conduct might be divisive, and that's a problem, but it's outside the scope of my response.

In the case of ML5, which you referenced, what they called "Code of Conduct" isn't truly a CoC in the traditional sense, in the same way the Contributor Covenant is.
While the word "Code of Conduct" tipically refers to rules for maintainers and developers, within the project, the ML5 rules are actually now part of the license, making it an ethical license.

This is not to be confused with simply adopting a Code of Conduct in the traditional sense.
I am not trying to support the idea of codes of conduct, but none of the arguments I presented could be used against any actual code of conduct, good or bad.

Even before ML5 switched license (which happened so recently their project still contains the old license file in the same position) the "Code of Conduct" was meant to apply to users and this has been the case for two years.
Again, that goes well outside of the scope of Code of Conducts in the traditional sense, such as the Contributor Covenant.

Such set of rules, before relicensing, while ML5 was under MIT license, wasn't in practice actually enforceable on the user.
Whether ML5 was ever truly Open Source is, to me, an open question and really depends on how we think about the threshold of software freedom. I'd argue the user must be provided all permissions equally morally and legally, as both aspect may be equally important for the user.

I think the best way of addressing ethics while keeping the advantages of Open Source is not by imposing rules trough the project itself, nor to ask users to accept conditions, nor to force distributors to distribute statements. Rather, Open Source gives developers and organizations "fame", so much so that attribution is required by almost all Open Source licenses and that it's one of the reason software is developed at all.

Being known is useful partly because it gives the known a voice, and this voice can be used outside that which strictly is the project itself, such as the website of the organization developing the project.

Doing this is not an obvious choice, mind you, and it's not right for every organization.
But any issue it could possibly cause, any argument in favour or against, is completely separate and orthogonal to software freedom, Open Source and anything of the sort.

So, as for your idea:
> After taking a few days to reflect and spoke with @b-g also, decided to just include in a prominent way something like a "artist statement" on the website / documentation of OpenDataCam for next release... no re licensing, no code of conduct.. keep it very simple as always..

I am not supporting this, nor opposing it.
Nor could I, because I am not nearly familiar enough neither with the project, nor with what the statement will say.

What I will say however is that none of the arguments I provided in my previous comment and in the Twitter thread could possibly remotely apply to this choice, as they only apply to licensing (the original idea of this thread). This would remain true if the project adopted a Code of Conduct in the traditional sense.

My principal concearn is licensing and I truly believe you made the best choice and I am thankful to @tdurand, @b-g and everyone else involved in the evaluation, as choices like this affect the global trend of an decisions.
As for what else to do (such as adding a mission statement), I am sure this will be clarified by the rest of the discussion.

munsterlander, 2021-11-03:
I don't know if I am allowed to chime in on this, but I have always found the Linux Professional Institute to have a pretty good Code of Conduct / Ethics model.  I used theirs as a basis for another project -  https://www.lpi.org/conduct.

vsaw, 2021-11-04:
@munsterlander of course you are invited to participate in any discussion. And thanks for the pointers!

munsterlander, 2021-11-05:
@vsaw  Ok, I will take a go on a rough draft ethics statement / code of conduct. Also, why doing some testing on the dev branch (still trying to get it working), I noticed NextJS now by default, adds the project into their anonymous data collection:  https://nextjs.org/telemetry.  This can be disabled by `npx next telemetry disable` in the root of the project, but should this be in the instructions or in the build script as the readme.md states:

> OpenDataCam never records any photo or video data. The system only saves surveyed meta-data, in particular the path an object moved or number of counted objects at a certain point. The novelty of OpenDataCam is, that everything happens on location, while no visual data is saved or sent to online cloud processing.

Do you think the inclusion of telemetry reporting to NextJS goes against this concept?