opendatacam/opendatacam #395: Unable to use terminal after installing OpenDataCam

ilsa16, 2021-05-12:
I have installed OpenDataCam with docker on Jetson Nano, and after closing and trying to reopen my LXTerminal, I cannot type or do anything in it. I have tried every shortcut that I found on the internet to reset it but still doesn't work (Ctrl+C, Ctrl+J, etc.).
!
[LXterminal](https://user-images.githubusercontent.com/74668436/117963648-8f517780-b318-11eb-8260-4e24d7d6a11e.png)

This is definitely an OpenDataCam issue because I had this issue happen on another SD card, after installing it and restarting the Jetson Nano. At first I thought it was an issue due to the auto-start feature of the software, but in this case, I even shut it down with "sudo docker-compose down".

Might be something really simple since I am inexperienced with docker containers and Linux. Any help would be greatly appreciated since I can't do anything with my Jetson Nano terminal atm, and formatting/reinstalling everything would be a pain. Thank you!

vsaw, 2021-05-12:
What output do you see when you type `docker-compose up´? Also what happens if you try `docker-compose up -d` to start the containers in the background?

ilsa16, 2021-05-12:
Thanks for the quick answer. The issue is I can't type anything in the terminal, it is basically unresponsive. It interacts with mouse movements, so it's probably not an OS issue. I was trying to find any process that might be causing the issue using Task Manager, but I'm not sure what might be causing it. Any ideas?

vsaw, 2021-05-12:
That is odd. First time I've heard something like this so I don't really have any idea what could be going on. But if you do find out please share it here. I'm curious.

ilsa16, 2021-05-12:
Is there any way that I can stop/restart docker without the console? Maybe a specific process on task manager?

secg-sr1, 2021-05-12:
Hi! The same here. I'm working with a jetson nano 2gb and i copied in another folder config files and delete them from from the main directory. It is not practical but i made that to have access to linux terminal again.

tdurand, 2021-05-13:
maybe jetson nano 2gb isn't powerfull enough for OpenDataCam and launching it takes all memory available, do you have a swap partition setup ?

I ordered one, I will receive in the coming days, will try to reproduce

ilsa16, 2021-05-13:
> Hi! The same here. I'm working with a jetson nano 2gb and i copied in another folder config files and delete them from from the main directory. It is not practical but i made that to have access to linux terminal again.

Great tip, it worked. Thank you!

ilsa16, 2021-05-13:
> maybe jetson nano 2gb isn't powerfull enough for OpenDataCam and launching it takes all memory available, do you have a swap partition setup ?
> 
> I ordered one, I will receive in the coming days, will try to reproduce

ilsa@host:~$ free -m
              total        used        free      shared  buff/cache   available
Mem:           1971        1038         134          48         798         796
Swap:         11225          10       11215

Not sure how ideal this is, but I am working with a 128GB SD card so I thought why not.

unrzn0, 2021-05-13:
This ticket came just in time :) Thinking about getting all hardware for OpenDataCam and the 2GB Jetson Nano looks from remote as if it would also work, as the only difference are the number of ports, WiFi, and the 2GB ram instead of 4GB? Or do I miss something? If anybody can give a thumbs up that the 2GB version works, it's in my shopping basket instantly (the 4GB version is double the price!)

secg-sr1, 2021-05-13:
Jetson nano 2gb real time takes all memory but work fine with uploaded videos. 
[image]

vsaw, 2021-05-13:
I believe @MaximilianSteverding is running ODC on a 2 GB Jetson Nano from Webcam. But I think it is running from Headless mode (no Desktop environment running on the device). Not sure if that makes a difference.

MaximilianSteverding, 2021-05-13:
I am running ODC on 2GB Jetson Nano and have successfully run it in the following setups:

1. with Jetpack and the docker image via file / raspberry cam - With this setup the framerate was about 9-12 fps. I did not do a long run but I had recordings of 2-3 hours running successfully on Jetson Nano 2GB. Nevertheless I did not monitored if the memory was running full.

2. balena deployment with external usb-webcam (free for up to 10 devices) - with this setup there was a significant performance boost with the framerate between 12-19 fps. I was able to perform a long run of 4 days straight with about 400.000 Tracker entries. After the 4 days the Jetson Nano froze. Nevertheless the CSV Data was not lost and could be recovered in the webinterface. Through the whole recording about 1.6GB of RAM was used. The CPU ist at about 50% all the time, going up to 70% when many objects needed to be traced. The only problem occurred when I tried to download the JSON file with the RAM running full and the CPU going max, ending with a server timeout.

In any setup I would recommend cooling (e.g. through a fan). This will keep the CPU at 60 degrees.

MaximilianSteverding, 2021-05-13:
@secg-sr1 are you running the Jetson Nano 2GB at 10 watts? In your screenshot it says "System throttled due to Overcurrent". You can change the power consumption from 5 to 10 watts (with the appropriate power source). This will also increase the performance and stability of the system.

secg-sr1, 2021-05-14:
Hi @MaximilianSteverding ! yes, I'm running the Jetson Nano 2GB at 10 watts. The warning pops up when running anything which takes all memory available. I'm really interested to try balena. Thanks for the advice and tips.

closed 2021-05-18 by ilsa16

vsaw, 2021-05-19:
@ilsa16 did you solve it?

aadibajpai, 2021-06-30:
I encountered the same issue, didn't have the prompt in lxterminal and wasn't sure if it was related to opendatacam. Found this thread: https://forums.developer.nvidia.com/t/system-update-breaks-terminal/170860 and followed it which fixed the issue. The author of that thread mentions running opendatacam as well. 

I switched to a virtual tty using ctrl+alt+f2 then ran `sudo chmod 666 /dev/pts/ptmx` then switched back to the session with ctrl+alt+f7 which fixed it.

IssaIssa1, 2021-11-29:
@aadibajpai Thank you. This solve the problem temporally but when I reboot the jetson I have again the same issue. Did anyone find a way to fix it once for all?

vsaw, 2021-12-08:
I am not aware of anyone. Could be that many people run OpenDataCam in headless mode and therefore never experienced the issue.

I am currently running Opendatacam on a 2 GB Jetson Nano (JP 4.6) in headless mode, and everything works as expected, the system is completely responsive in SSH.

IssaIssa1, 2021-12-09:
Im only getting this problem when I shutdown the jetson. I have two questions, if you dont mind. Have you got such error if you reboot the jetson remotely? And what are you using for remote access [I am trying to set up vnc from windows to ubuntu (jetson) but it's not working well ]?

vsaw, 2021-12-09:
I have never encountered that problem as I do not run a graphic user interface on the Nano. I Setup the device using headless install and access it using SSH only.

munsterlander, 2021-12-09:
Like VSaw, I use only headless mode doing exactly as NVidia has in their getting started.  I use Putty on Windows 10 connecting to the Jetson running their flavor of Ubuntu.  I have not done anything special and am running the bare minimum install from their docs without issue.

AlexHolly, 2022-11-03:
I ran into the same issue.

As a workaround I added the command `chmod 666 /dev/pts/ptmx` to a service which is executed on boot.
Make sure to remove the setting "User" to run as root.
(https://medium.com/@benmorel/creating-a-linux-service-with-systemd-611b5c8b91d6)

Related links:
https://askubuntu.com/questions/1305349/terminal-error-there-was-an-error-creating-the-child-process-for-this-terminal

https://forums.balena.io/t/udev-1-breaks-host-ubuntu-terminals/40619
https://hub.docker.com/layers/opendatacam/opendatacam-darknet-base/v1.0.1-odc-Yolo-v3-nano/images/sha256-b36b3cb4db5fa5e5e6aa14bcdb4572120240cff36d4eaef6bf7490667cbee138?context=explore

https://github.com/kubernetes/kubernetes/issues/18230