opendatacam/opendatacam #398: [Hardware] Clusters of Jetons

tdurand, 2021-05-16:
To mention it exists in the documentation.

https://wiki.seeedstudio.com/Jetson-Mate/

vsaw, 2021-05-17:
Does this have any use besides setting up a Kubernetes cluster? Seems to me like it's of little use if all you care about is getting more FPS from Darknet.

tdurand, 2021-05-17:
honestly I didn't review it, you think it is not able to get for example 2 or 3 nano together to have same FPS as 1 xavier ? but yes even so I agree it is useless as you are better off buying one xavier NX.

vsaw, 2021-05-17:
I only took a brief look at it. And as far as I can tell it seems to be basically just a Ethernet connection between the Jetsons, with each Jetson running their own OS and SW stack. So unless you have an application that benefits greatly from a Kubernetes cluster you will not receive any performance gains.

tdurand, 2021-05-17:
okay, thanks for looking into it ! another thing that was eye candy ! 😂

closed 2021-05-17 by tdurand