opendatacam/opendatacam #400: Changing time in export.csv

P-Cano, 2021-05-19:
Hello

I would like to ask if there was a way to change the time exported in the "export.csv" file in order to display my local time instead of UTC. Can you help me out? 

There´s a closed issue with this exact topic but as for this version, I believe that no longer applies.

Kind regards

vsaw, 2021-05-20:
There is no built in way. You would have to convert the time zone in your Spreadsheets or whatever program you use for looking at the CSV. Sorry

tdurand, 2021-05-22:
you would need to change this line of code: https://github.com/opendatacam/opendatacam/blob/development/server.js#L889

closed 2021-05-22 by tdurand

P-Cano, 2021-05-26:
> you would need to change this line of code: https://github.com/opendatacam/opendatacam/blob/development/server.js#L889

Thanks a lot @tdurand , that helped a lot

BTW I edited that line of code with:

"timestamp: countedItem.timestamp.toLocaleString('es-MX', { timeZone: 'America/Mexico_City' }),"

Thanks again. Regards