opendatacam/opendatacam #401: Question: Where is streaming code?

kyasby, 2021-05-21:
I would like to know where in the codes the value  "YOUR IP CAM STREAM (can be .m3u8, MJPEG ...), anything supported by opencv", is provided to gstream. of the key "remote_cam" is loaded.

Thanks.

haviduck, 2021-05-21:
not sure if i understand correctly, do you mean where the rtsp stream is being used? if so its set as a parameter when they do their exec darknet -i detector demo line. dont remember the exact js but a quick search will get you there

tdurand, 2021-05-22:
exactly here: https://github.com/opendatacam/opendatacam/blob/development/server/processes/YoloDarknet.js#L57

We construct a bash command like:

```
./darknet detector demo cfg/coco.data cfg/yolov4-tiny.cfg yolov4-tiny.weights "v4l2src ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink" -ext_output -dont_show
```

Darknet gets in input the Gstreamer pipeline and feeds it to OpenCV

tdurand, 2021-05-22:
feel free to re-open if you need more info

closed 2021-05-22 by tdurand