opendatacam/opendatacam #402: OpenDataCam v4 public roadmap

tdurand, 2021-05-22:
I took some time to do issue triage and create a milestone for v4 release and I created a "OpenDataCam v4" project board also: https://github.com/opendatacam/opendatacam/projects/3 , not sure if it is useful, this is a try

The very short summary of the next version is:

- update the website: integrating documentation into it (idea is not to scare non tech savy people with a README only on github): https://github.com/opendatacam/opendatacam-website/issues/21 , with 3 call to action on the landing page: "Try it on Android", "Try in on Jetson" , "See documentation" , with a 1 min video featuring the UI 

- work on a section showing all the different use cases: https://github.com/opendatacam/opendatacam/issues/84 👀)

- add a "ethical statement" in a prominent way , something like 2 paragraph we will write as a community, no re-licensing just keep MIT all the same ..  https://github.com/opendatacam/opendatacam/issues/390

- Update every dependencies to make sure that ODC works with: latest jetpack version, latest darknet (scaled yolo weights also), docker images etc etc..

- Fix stuff that are quick wins

- showcase contributors better: https://github.com/opendatacam/opendatacam/issues/120

- + add anything you would like contribute that makes sense for the project ✌

The planning for this is to release v4 version by mid-July , I will be actively working on this in June

vsaw, 2021-05-22:
How difficult would it be to merge the Android and Jetson forks?

tdurand, 2021-05-22:
it is not super simple, diff is documented here: https://github.com/opendatacam/opendatacam-mobile/blob/master/DEVELOPER_NOTES.md#documented-diff-between-main-repo-for-the-nodejs--nextjs-part  .. some stuff would be tricky to support both.. (not impossible though, just takes time to test and dev)

I will not have the time to do it for v4 release.. but I'm happy to keep this "mobile" branch, and document a bit more how we pull the opendatacam code into the android app (do an architecture schema).

My approach is to see if the Android app is of any use for people, and if it gets traction let's later on work on having exactly the same codebase

tdurand, 2021-05-22:
> I think it would make all our life's easier if we can avoid maintaining 2 forks.

My approach is we don't maintain the Android code, if we want to fix stuff for the Android app, first we do the job to merge the codebase.. right now just consider the https://github.com/opendatacam/opendatacam-mobile as a prototype we don't provide support

tdurand, 2021-08-11:
Jetpack 4.6 out: https://www.jetsonhacks.com/2021/08/10/jetpack-4-6-production-release/