opendatacam/opendatacam #405: Add Support for Xavier NX

vsaw, 2021-05-24:
I know it works. We just need to add it to the release and documentation.

azrin1972, 2021-05-30:
I'm trying to install opendatacam without docker. I started with compiling darknet. but I received and error as follows

`./src/convolutional_layer.c:286:24: error: ‘CUDNN_CONVOLUTION_FWD_PREFER_FASTEST’ undeclared (first use in this function); did you mean ‘CUDNN_CONVOLUTION_BWD_FILTER_ALGO_3’?
     int forward_algo = CUDNN_CONVOLUTION_FWD_PREFER_FASTEST;
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                        CUDNN_CONVOLUTION_BWD_FILTER_ALGO_3
compilation terminated due to -Wfatal-errors.
Makefile:162: recipe for target 'obj/convolutional_layer.o' failed
make: *** [obj/convolutional_layer.o] Error 1`

I've followed the how to line by line. Please advise

vsaw, 2021-05-30:
Try using upstream darknet. The ODC fork right now is only used for the docker image and no longer compiles with the current JetPack.

azrin1972, 2021-05-30:
I'm sorry for my little knowledge - how do I do this actually?

> Try using upstream darknet. The ODC fork right now is only used for the docker image and no longer compiles with the current JetPack.

vsaw, 2021-05-30:
> I'm sorry for my little knowledge - how do I do this actually?
> 
> > Try using upstream darknet. The ODC fork right now is only used for the docker image and no longer compiles with the current JetPack.
> 
> this is what I actually did
> 
> `git clone https://github.com/AlexeyAB/darknet.git`

This looks correct. Then follow the instructions from https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md for Jetson Xavier

> and for your Info, I'm using Jetson Xavier NX and I couldn't find a way to install JetPack 4.3 as suggested, I'm using 4.5.1 instead

That is fine I'm also using JetPack 4.5. As far as I remember the release notes 4.5.1 should work fine though as well. (JetPack 4.3 is much older than the Xavier NX, this is why you can only run JetPack 4.5.1 on it.)

azrin1972, 2021-05-31:
ok got it to work but the the FPS is on average 4.5 only 

[image]

vsaw, 2021-05-31:
Try running Yolo Tiny. I get up to 60 FPS there. 🙂

azrin1972, 2021-06-01:
after a few hours of analytics, HTOP shows significant amount of memory usage and everything slows down. I think I'm trying the deepstream instead of darknet.

will update the findings soon

vsaw, 2021-06-01:
Do you know if it's Darknet or the ODC node app?

Also DeepStream is not yet officially supported but there are a few community efforts that try to do this. And any contribution would of course be greatly appreciated.

azrin1972, 2021-06-01:
I'm not sure if it's Darknet or ODC. I saw there's a repo on Deepstream that I would like to try. Because my plan is to use with at least 4 streams of IP Camera per Xavier NX. But as per today's test at only avg 10FPS - I need to continue exploring :)

vsaw, 2023-09-20:
Xavier NX uses the same code and binaries as the "original Xavier". Therefore once https://github.com/opendatacam/darknet/pull/23 is merged, ODC officially supports Xavier NX devices as well!

closed 2023-10-19 by vsaw