opendatacam/opendatacam #408: Parameters to set FPS in video input file

nizaet, 2021-06-10:
Hello everyone,

I'm using opendatacam on desktop with gtx1080 gpu.
I do drag and drop files in the browser, then the video runs at 35-40 Fps.

how to set the config setting to change the Fps in 20-25 fps in video input file

Thank you

tdurand, 2021-06-14:
Hi, 

This is not possible currently

In order to implement this, would need to set a limit into the darknet code to process the frame not more than 30 FPS .. 

Not planning to add this for now, but you are welcome to do so and contribute a pull request

Have a nice day

closed 2021-06-14 by tdurand

vsaw, 2021-06-14:
Another possible work around could be: Upload the file onto the machine using SSH and then use the gst [filesource](https://gstreamer.freedesktop.org/documentation/coreelements/filesrc.html?gi-language=c#filesrc) and [videorate](https://gstreamer.freedesktop.org/documentation/videorate/index.html?gi-language=c) filters