opendatacam/opendatacam #410: Different box representation in tracker

akretz, 2021-06-24:
Hello everyone,

OpenDataCam uses the `x`, `y` properties of detection boxes to refer to the center point, while the computations used in [node-moving-things-tracker](/opendatacam/node-moving-things-tracker) use the top left corner as reference point instead. This results in a wrong IoU being calculated. This issue becomes apparent when you set the `iouLimit` to some high value and then the tracker fails a lot more often than when you compute the correct IoU. The solution would be to either convert between both representations when passing data between ODC and the tracker or to use the same representation everywhere. In the latter case, the following lines would need to be modified to use the center point as reference:

https://github.com/opendatacam/node-moving-things-tracker/blob/474f50a4b9c9fdaba5e2bc89085498c0b6848445/utils.js#L10-L13
https://github.com/opendatacam/node-moving-things-tracker/blob/474f50a4b9c9fdaba5e2bc89085498c0b6848445/utils.js#L38-L41
https://github.com/opendatacam/node-moving-things-tracker/blob/474f50a4b9c9fdaba5e2bc89085498c0b6848445/main.js#L139-L140

vsaw, 2021-06-24:
Good catch! Looking at the [node-moving-things README](https://github.com/opendatacam/node-moving-things-tracker#detections-input-and-tracker-output-format) I believe it's actually a bug in the implementation and `x`,`y` should refer to the center point there as well!

At least this way it will be consistent with [Darknet](https://github.com/opendatacam/darknet)

Would you be okay if we create an issue in the node-moving-things-tracker to keep track and implement the fixes? I would then rename this issue to keep track of updating the node-moving-things dependency once it is fixed.

akretz, 2021-06-24:
I've just created a pull request which should fix the issue. I'm not sure if I've missed anything, but I've ran the MOT17 benchmark on the fix and it performs almost the same. The small differences are probably due to the velocity vector being computed with respect to the center points rather than the top left corners.

> Would you be okay if we create an issue in the node-moving-things-tracker to keep track and implement the fixes? I would then rename this issue to keep track of updating the node-moving-things dependency once it is fixed.

Yes, sure!

vsaw, 2021-06-24:
I've created a new issue in https://github.com/opendatacam/node-moving-things-tracker/issues/24 to track the implementation.

This issue will be renamed to remind us to update the node-moving-things depedancy in Opendatacam.

vsaw, 2021-06-25:
Sorry closed by mistake

vsaw, 2021-06-25:
I could successfully run the following docker image with node-moving-things-tracker@0.8.1 https://hub.docker.com/layers/vsawdoteu/opendatacam/development-nano/images/sha256-f2278fac7d7a252747169bdebb7081d22086b45dcda9f8f1f5164886f6c2e22c?context=explore

closed 2021-06-25 by vsaw

tdurand, 2021-06-25:
❤