opendatacam/opendatacam #411: Can this run on mp4?

cedricve, 2021-06-24:
Hello we are wondering if this could run directly on mp4 files as well?

Kind regards
Cedric

vsaw, 2021-06-24:
Yes it can. Simplest way is to "drag n drop" a video file on the web interface. OpenDataCam will restart and start running your file.

cedricve, 2021-06-24:
that's cool! and is there anyway to send them to an API, and let opencam process it in the background?

vsaw, 2021-06-24:
Yes. The endpoint is not documented but it's POST to /files

See the implementation in server.js

Rest of the API is documented here https://opendatacam.github.io/opendatacam/apidoc/

cedricve, 2021-06-24:
Ok cool we might write an article about this for integrating with kerberos.io <http://kerberos.io/> ;)

vsaw, 2021-06-24:
Great! Drop us a note when it's ready :-)

closed 2021-06-24 by vsaw