opendatacam/opendatacam #413: Speed estimation for distance between two lines

a2moosh, 2021-06-26:
If for example two lines are defined with a distance of x meters from one to the other it would be great if there could be a speed estimation added to objects.

vsaw, 2021-06-27:
You are right that should give you an approximate speed. However it is currently not part of OpenDataCam and would either have to be implemented (feel free to give it a go 🙂) or you will have to do the post-processing outside of OpenDataCam.

vsaw, 2021-06-27:
@tdurand Ive renamed to issue and will leave it open for people to see and measure interest. (I get the feeling it pops up every couple of months)

haviduck, 2021-09-19:
well, i think this is doable with the current api. last time i played with odc i remember seeing drawable areas for counting. that plus counterhistory where you get entry and exit timestamps is enough along with a certainty of fps and knowing the realworld box distance.

from there on out its just math. but accurate estimations are really sketchy.. youd have to place your roi in a head on view. imagine doing this in a roundabout. youd have to calculate curvature etc..  that being said, api is enough unless i read the docs completely wrong here

LucasCanete, 2025-02-05:
@a2moosh @vsaw I have already made a python implementation to fill this gap. Speed can be measured between two lines and the data will be uploaded to Opendatacam's built in Mongo DB. Apart from that the user can define the upload rate, can measure in more that one area at a time, etc. Take a look: https://github.com/LucasCanete/OpenDataCam_SpeedDetection/tree/main