opendatacam/opendatacam #414: detection and dropped video are not in sync

zeljkomarinkovic, 2021-07-01:
[image]

ODC installation without docker, running on desktop PC with ubuntu 20.04 installed and AlexeyAB/darknet

ODC is running on localhost and port is forwarded via ngrok so we can access it from anywhere. 

Files are dropped from my local pc which is on same network as machine running ODC

There is quite big missmatch between detection and video in background.

vsaw, 2021-07-01:
This is known, and as far as we can tell only cosmetic (in the sense that detections and counting should still work as expected).

I don't really know a quick way to fix this except refreshing the page browser window, which re-syncs the stream.

tdurand, 2021-07-02:
also try to reduce the resolution of the camera, high res stream have the tendency to get out of sync.. yolo only process at his best 600x600 pixels

closed 2021-07-27 by tdurand