opendatacam/opendatacam #415: Fresh Install instructions fail at installing docker-compose

napking, 2021-07-07:
Quick background: I've been doing a fresh install on my Jetson Nano and updating documentation at the same time. Over the past few days and attempts, I have failed at [installing docker-compose](https://github.com/napking/opendatacam/blob/development/documentation/jetson/JETSON_NANO.md#4-install-opendatacam)

Terminal output:
```
Collecting cryptography>=2.5 (from paramiko>=2.4.2; extra == "ssh"->docker[ssh]<4.0,>=3.7.0->docker-compose==1.24.0)
  Using cached https://files.pythonhosted.org/packages/9b/77/461087a514d2e8ece1c975d8216bc03f7048e6090c5166bc34115afdaa53/cryptography-3.4.7.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-6hzuh9y4/cryptography/setup.py", line 14, in <module>
        from setuptools_rust import RustExtension
    ModuleNotFoundError: No module named 'setuptools_rust'
    
            =============================DEBUG ASSISTANCE==========================
            If you are seeing an error here please try the following to
            successfully install cryptography:
    
            Upgrade to the latest pip and try again. This will fix errors for most
            users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
            =============================DEBUG ASSISTANCE==========================
    
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-6hzuh9y4/cryptography/
```

many searching around, seems that updating pip is required, specifically the setup-tools module. Not just any update command was working either.
For anyone else running into this issue, here is the command that had worked for me:
`python3 -m pip install -U pip setuptools`

need to fully replicate by using this command directly before `pip3 install docker-compose`
need to replicate on other systems

Relevant links:
[cryptography#5753](https://github.com/pyca/cryptography/issues/5753#issuecomment-822071521)
[azure/azure-cli#16858](https://github.com/Azure/azure-cli/issues/16858#issue-803817486)

vsaw, 2021-07-08:
Thanks. I guess those need to be updated. 

Which JetPack version are you using?

napking, 2021-07-08:
Failed to install docker-compose in both 4.3 and 4.5 jetpack. I only managed to do my troubleshooting in 4.5. I don't think that would have had an issue with pulling python installs, but I'm not an expert.

vsaw, 2021-07-12:
Hmm and if you try to install docker-compose via `apt`? At least that's what I do and it all worked so far. (I believe the documentation is a bit outdated and may have been written before there was proper support for `docker-compose`

napking, 2021-07-15:
Confirmed with a fresh install. Updated my own development branch with the correct install commands.

vsaw, 2021-07-16:
@napking Great!

> Updated my own development branch with the correct install commands.

Would you mind setting up a quick Pull request? I'd love to have this fixed in upstream ODC as well. 🙂

vsaw, 2021-07-16:
Btw. I re-opened the issue as a reminder that the documentation needs fixing :-)

vsaw, 2021-09-16:
Closed as v3.0.2 is now released.

See also https://github.com/opendatacam/opendatacam/releases/tag/v3.0.2

closed 2021-09-16 by vsaw