opendatacam/opendatacam #416: Create Docker build for Jetson TX2

IrishTrafficSurveysDev, 2021-07-12:
Hey,
According to the quick setup docs:
https://github.com/opendatacam/opendatacam#2-install-and-start-opendatacam-

Docker build for Jetson TX2 isn't available and needs to be installed without docker. 
Would it be possible to create a Docker build for Jetson TX2?

vsaw, 2021-07-12:
The documentation might be getting a bit out of date. Have you tried running the Jetson nano docker image on the TX2? Could probably just work, at least I don't see a reason why not.

tdurand, 2021-07-22:
Hi, 

Due to too much maintenance work (it's not just building it's testing etc etc, also needed to build on the target device at the time), decided not to do docker images for TX2 .. if you use the nano one if should work but performance won't be the best... (due to some CUDA arch flag compiling darknet)

You can build you own, the only thing that need to be changed from jetson nano is this line for compiling darknet: https://github.com/opendatacam/opendatacam/blob/master/docker/build/nano/Makefile#L45

Will see with @vsaw work for v4 if auto build of docker image for all jetson type of device is possible

good luck

closed 2021-07-22 by tdurand

IrishTrafficSurveysDev, 2021-07-23:
Just for anyone reading this, I can say that I did manage to create a docker build for the TX2, and performance was quite good. Thanks for the help @tdurand and @vsaw !

vsaw, 2021-07-23:
Great. Btw if you want, you can add your TX2 docker file as a pull request. Would be greatly appreciated. We are currently building automatic development previous of the docker containers vie GitHub Actions. You can find them on Docker Hub, they start with `development-`

In the future we are also investigating automatically creating docker builds for all new releases. 

If you have questions on how to do so feel free to ask me any time