opendatacam/opendatacam #417: Connection refused ?

damgaarderik, 2021-07-13:
I have installed opendatacam on ḿy laptop, running ubuntu 20.04 amd64

Everything seems to install and running correctly but when I start opendatacam it is looking for an IP-address that is refused (other than the localhost)

[image]

Is there any settings or something that I am missing ?

vsaw, 2021-07-14:
Assuming Darknet is running this is most likely a known bug on using ODC with NAT/Firewalls. See this issue provides a workaround for you https://github.com/opendatacam/opendatacam/issues/374

If so I would consider this a duplicate and close this issue.

damgaarderik, 2021-07-14:
I have followed the guide, using a docker image. Unfortunately I'm not that skilled to understand the workaround? 

-    Setup port forwarding for 8070 and 8090 (not preferred)
-  Fix the Darknet hostname in https://github.com/opendatacam/opendatacam/blob/development/server/Opendatacam.js#L792 and https://github.com/opendatacam/opendatacam/blob/development/server.js#L188 to localhost

==> I have no idea where to eg. find Opendatacam.js and/or server.js to tweak the IP in a docker-container unless the fix is that I download the sourcecode and build  ?

akretz, 2021-07-14:
I believe OpenDataCam tries to connect to Darknet using the same hostname you're using to access the web interface. It appears that you're accessing the interface via http://172.18.0.3:8080. Have you tried using http://localhost:8080 instead?

damgaarderik, 2021-07-15:
Hi akretz,
I do not see any place (eg. in config.json) where I can specify the IP-adresses, seems to be handled by the code, to try to resolve to a ip-address?
I have no ip-address of 172.18.0.3, so I´m not sure how opendatacam resolve to this ip-adress ???
I have installed ocd on a local laptop (to test it) so I guess it should just see localhost ?

damgaarderik, 2021-07-15:
ok I can see that docker actually creates a inet :+1: 

docker0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        **inet 172.17.0.1**  netmask 255.255.0.0  broadcast 172.17.255.255
        ether 02:42:a1:c6:0d:e0  txqueuelen 0  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

So I guess that ODC in some way get an IP-address based on docker ip-adresses ???

vsaw, 2021-07-15:
@damgaarderik Sorry, the workaround I sent you would require you to run ODC from source.

@akretz is correct in his assumption that ODC tries to connect to darknet via the HTTP host that you put in your browser to connect to the ODC web interface.

That being said: @damgaarderik could you please share a few things with us

1. What address do you type in your browser to connect  to ODC?
2. Are you behind a NAT or do you have any other special network setup? (e.g. Firewall)
3. What ports have you set up in your dockerfile?

The workaround so far still is either running (and patching) ODC from source, or opening up the ports 8070 and 8090.

akretz, 2021-07-15:
> I do not see any place (eg. in config.json) where I can specify the IP-adresses, seems to be handled by the code, to try to resolve to a ip-address?

I meant to access ODC in the browser via http://localhost:8080 instead of http://172.18.0.3:8080.

My theory of what is happening is that the connection attempt to 172.18.0.3 gets routed to the host and the host doesn't forward it back to the container, because IP forwarding is disabled on Ubuntu by default. I'd suggest trying the following:

- Use only http://localhost:8080 to access ODC in the browser after you've started ODC.
- Enable IP forwarding in your host. That can be done with
`sudo sysctl net.ipv4.conf.all.forwarding=1`
and then
`sudo iptables -P FORWARD ACCEPT`
just to make sure forwarding is not blocked by iptables.

vsaw, 2021-07-15:
> My theory of what is happening is that the connection attempt to 172.18.0.3 gets routed to the host and the host doesn't forward it back to the container, because IP forwarding is disabled on Ubuntu by default.

Possible but unlikely if docker-compose is used correctly as that should take care of the IP-tables. Which is another question for @damgaarderik: What command do you use to start ODC?

damgaarderik, 2021-07-16:
Ok I have enabled IP forwarding as described.
I start the container with the following command "sudo docker-compose up" this generate the output from ODC. 

checking IP 's  : 
sudo docker network inspect -f '{{json .Containers}}' 4b9608fe20b3 | jq '.[] | .Name + ":" + .IPv4Address'
"desktop_opendatacam_1:172.18.0.3/16"
"desktop_mongo_1:172.18.0.2/16"

localhost:8080 and 172.80.0.3:8080 gives the same result :  
[image]

damgaarderik, 2021-07-16:
[image]

should it be possible to access something on port 8070 and 8090, eg. from the browser ? eg. 172.18.0.3:8070 ? There is no response.

akretz, 2021-07-16:
> should it be possible to access something on port 8070 and 8090, eg. from the browser ? eg. 172.18.0.3:8070 ? There is no response.

Yes, you should be able to connect there via the browser and see the stream of JSON objects and the video stream.

[image]

I interpret this output as Darknet crashing early. So perhaps it's not a network issue.

damgaarderik, 2021-07-16:
I have installed OCD on a local laptop running GTX3070, AMD processor.

I have followed the instructions on : https://github.com/opendatacam/opendatacam#for-desktop-machine-nvidia-container-toolkit-

then 
 Install command for a Desktop machine
# NB: Will run from demo file, you can change this after install, see "5. Customize OpenDataCam"
./install-opendatacam.sh --platform desktop

which creates an image for docker an starts it.
then I enter localhost:8080 in the browser .....

[image]

but it never get any further than the initialization-page, where 
I guess that darknet or something is not working, because the error messages i actually : 
[image]

which indicates that Yolo is suffering after some while ?44
[image]

vsaw, 2022-02-14:
I assume/hope this is now solved as https://github.com/opendatacam/opendatacam/pull/527 is merged in development.

closed 2022-02-14 by vsaw