opendatacam/opendatacam #421: Mongodb Maintience

Masonzton, 2021-07-20:
I am running opendatacam on a Jetson Nano.  It had been running for quite some time.  As a result I believe that the mongo database is way too big.  The disk usage of the jetson was at 100% (64GB sd card) and purging the mongo container and deleting the dangling volumes brought it down to around 56%.  Looking at the mongo db through the command line at a latter time, the tracker collection is up to 12GB.  The first thing I found to fix this is to do capped collections. Is there a clean way of doing capped collection for a mongo database docker?  I saw this [issue](https://github.com/opendatacam/opendatacam/issues/387) where someone mentioned converting to capped collections through the terminal and that does seem to work, however it is involved.  I would like for this to happen without user intervention, especially if you need to completely reload the docker image.  I would rather not edit the source code to use capped collections since I may want to update to v4 soon.  How are people managing the size of their Mongo Database?  The current idea I am working on is to have some script that converts the tracker collection to capped.  The other idea is to clean and purge mongo if disk space gets too tight.  Is there a better way?  Is editing and recompiling the source code not as tedious as I think it is?

Side note: I am doing a new recording every 60 seconds with node red to count how many people cross a line, which may be part of my problem.  Also the frame is quite busy.

vsaw, 2021-07-21:
True Development version already only persists tracker data if there are object in the frame, which depending on your usage could already be a significant improvement. 

Otherwise I will try to implement https://github.com/opendatacam/opendatacam/issues/388 as part of the upcoming v4 release. 

Capped data collections so far have not been considered, and I'm generally interested in exploring this option. But I would like to understand first how people use the tracker data (if at all)

parthapp, 2021-11-12:
Hello Sir, I can provide a couple use cases for tracker data. Suppose I am recording an intersection and all approaches are visible. If I can track vehicles from the time they leave the stop bar and then cross the bar for their departure direction, I can process this data to generate turning movement counts at an intersection, by time of day. This information is critical for traffic signal operations to determine optimal splits for a signalized intersection. This data is also critical in conducting traffic signal warrant studies, essentially data collection that serves as a basis to determine if an non-signalized  (2 way/4 way stops) intersection warrants a traffic signal. Tracker data could also be used to place 2 lines a known distance apart, track the passage of each ID. Distance apart/(time difference between passage) = speed.

vsaw, 2022-02-13:
Persistance of the tracker can now be controlled in development with the merge of #525 

I will therefore consider this issue fixed for now. Please reopen if you disagree :-)

closed 2022-02-13 by vsaw