opendatacam/opendatacam #422: OpenDataCam with DeepStreamv5 - Instructions?

IrishTrafficSurveysDev, 2021-07-23:
Hey there  👋 ,
Just happened to stumble upon this repo:
 https://github.com/opendatacam/opendatacam-deepstream.
I really like the idea of integrating deepstream with OpenDataCam, however I am not sure whether the documentation [here](https://github.com/AtosInnovationBordeaux/opendatacam-deepstream/tree/master/deepstream_patch) is still applicable to Deepstream v5. 

I would like to know whether or not to follow the same patch instructions as written in this documentation, or if the instructions would be different. I would love to know so that I can document my path to integration and create a PR.

Thanks 😄

vsaw, 2021-07-23:
The DeepStream port has been unattended for some time and most likely the instructions need to be updated. 

If you are willing to do the effort please go ahead. Ideally though you can merge the changes in the main repository as I would love in the end to offer "pluggable" neural network engines for ODC instead of maintaining different forks for different engines.

tdurand, 2021-07-27:
Yeah, the deepstream fork is outdated, maybe some info can be useful for integrating with deepstream v5 though.

You have also this thread with some interesting discussion about this: https://github.com/opendatacam/opendatacam/issues/112

Some people did it successfully but so far nothing usable "out of the box" was really made public

good luck ! closing this here, maybe the best if you start to work on this is to continue the discussion here: https://github.com/opendatacam/opendatacam/issues/112

closed 2021-07-27 by tdurand