opendatacam/opendatacam #423: Data page shows Counter and Tracker while counting, but goes blank after counting [Jetson Nano, JetPack 4.3]

kyutums, 2021-08-08:
I installed ODC 3.0.1 on a Jetson Nano via the Docker method. When I access ODC via http://192.168.0.11:8080/ and upload a video file to count the vehicles, it seems to work. However, the Data tab / area is blank at the beginning, has content during Recording, and goes blank again after the recording. I thus cannot download the data.

The counts from the Counter view seem to be correct though. The Console gives a ton of errors like the following:
`(node:119) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 6549)
(node:119) UnhandledPromiseRejectionWarning: MongoServerSelectionError: getaddrinfo EAI_AGAIN mongo
    at Timeout._onTimeout (/var/local/opendatacam/node_modules/mongodb/lib/core/sdam/topology.js:430:30)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)`

Should I download the data midstream, even if the recording is still "Ongoing", so I can download the data?

vsaw, 2021-08-08:
The error indicates that ODC can not connect to the Mongo DB. Can you confirm that ODC is running and that you have configured the right address?

If you don't know how to do this I can assist if you provide the following 🙂:

1. Output of `docker ps ` command while Opendatacam is running
2. Your `config.json` file
3. Your `docker-compose.yml` file
4. The command you use to start ODC

kyutums, 2021-08-09:
Yup, ODC is running (the web interface with the video of cars moving are showing). As far as I know (and I could be wrong), I have configured the right address. :)

Output of docker ps:

> CONTAINER ID   IMAGE                                 COMMAND                  CREATED        STATUS                            PORTS                                                                                                                             NAMES
> 8406f48f1e1b   opendatacam/opendatacam:v3.0.1-nano   "/usr/bin/entry.sh .…"   10 hours ago   Up 10 hours                       0.0.0.0:8070->8070/tcp, :::8070->8070/tcp, 0.0.0.0:8080->8080/tcp, :::8080->8080/tcp, 0.0.0.0:8090->8090/tcp, :::8090->8090/tcp   opendatacamnano_opendatacam_1
> 38c7bcddcadc   mongo                                 "docker-entrypoint.s…"   10 hours ago   Restarting (132) 26 seconds ago                                                                                                                                     opendatacamnano_mongo_1

config.json file
> {
  "OPENDATACAM_VERSION": "3.0.1",
  "PATH_TO_YOLO_DARKNET" : "/var/local/darknet",
  "VIDEO_INPUT": "file",
  "NEURAL_NETWORK": "yolov4-tiny",
  "VIDEO_INPUTS_PARAMS": {
    "file": "opendatacam_videos/demo.mp4",
    "usbcam": "v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink",
    "raspberrycam": "nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink",
    "remote_cam": "YOUR IP CAM STREAM (can be .m3u8, MJPEG ...), anything supported by opencv",
    "remote_hls_gstreamer": "souphttpsrc location=http://YOUR_HLSSTREAM_URL_HERE.m3u8 ! hlsdemux ! decodebin ! videoconvert ! videoscale ! appsink"
  },
  "TRACKER_SETTINGS": {
    "objectMaxAreaInPercentageOfFrame": 80,
    "confidence_threshold": 0.2,
    "iouLimit": 0.05,
    "unMatchedFrameTolerance": 5
  },
  "COUNTER_SETTINGS": {
    "minAngleWithCountingLineThreshold": 5,
    "computeTrajectoryBasedOnNbOfPastFrame": 5
  },
  "VALID_CLASSES": ["*"],
  "DISPLAY_CLASSES": [
    { "class": "bicycle", "hexcode": "1F6B2"},
    { "class": "person", "hexcode": "1F6B6"},
    { "class": "truck", "hexcode": "1F69B"},
    { "class": "motorbike", "hexcode": "1F6F5"},
    { "class": "car", "hexcode": "1F697"},
    { "class": "bus", "hexcode": "1F68C"}
  ],
  "PATHFINDER_COLORS": [
    "#1f77b4",
    "#ff7f0e",
    "#2ca02c",
    "#d62728",
    "#9467bd",
    "#8c564b",
    "#e377c2",
    "#7f7f7f",
    "#bcbd22",
    "#17becf"
  ],
  "COUNTER_COLORS": {
    "yellow": "#FFE700",
    "turquoise": "#A3FFF4",
    "green": "#a0f17f",
    "purple": "#d070f0",
    "red": "#AB4435"
  },
  "NEURAL_NETWORK_PARAMS": {
    "yolov4": {
      "data": "cfg/coco.data",
      "cfg": "cfg/yolov4-416x416.cfg",
      "weights": "yolov4.weights"
    },
    "yolov4-tiny": {
      "data": "cfg/coco.data",
      "cfg": "cfg/yolov4-tiny.cfg",
      "weights": "yolov4-tiny.weights"
    }
  },
  "TRACKER_ACCURACY_DISPLAY": {
    "nbFrameBuffer": 300,
    "settings": {
      "radius": 3.1,
      "blur": 6.2,
      "step": 0.1,
      "gradient": {
        "0.4":"orange",
        "1":"red"
      },
      "canvasResolutionFactor": 0.1
    }
  },
  "MONGODB_URL": "mongodb://mongo:27017",
  "PORTS": {
    "app": 8080,
    "darknet_json_stream": 8070,
    "darknet_mjpeg_stream": 8090
  }
}

docker-compose.yml
> version: "2.3"
services:
  opendatacam:
    restart: always
    image: opendatacam/opendatacam:v3.0.1-nano
    privileged: true
    volumes:
      - './config.json:/var/local/opendatacam/config.json'
    ports:
      - "8080:8080"
      - "8070:8070"
      - "8090:8090"
  mongo:
    image: mongo
    restart: always
    ports:
      - "27017:27017"
    volumes:
      - mongodb_data_container:/data/db
volumes:
  mongodb_data_container:

I don't use any specific command to start ODC. Whenever I restart my Nano, it ODC automatically starts with it.

vsaw, 2021-08-09:
It looks like the mongo service is not running. The docker status says it's restarting. Could you provide the logs of the mongo service to figure out why?

kyutums, 2021-08-09:
To clarify: I have to enter the container to check Mongo's logs. Did I
understand that correctly? (So sorry; newbie with Docker)

kyutums, 2021-08-10:
I tried viewing the logs at `/var/lib/docker/containers/[container-id]/[container-id]-json.log` and it was empty.

kyutums, 2021-08-10:
I did a `sudo docker ps` to check on the status of the containers and I noticed that the mongo containers was restarting.
```

CONTAINER ID   IMAGE                                 COMMAND                  CREATED        STATUS                           PORTS                                                                                                                             NAMES
8406f48f1e1b   opendatacam/opendatacam:v3.0.1-nano   "/usr/bin/entry.sh .…"   40 hours ago   Up 26 hours                      0.0.0.0:8070->8070/tcp, :::8070->8070/tcp, 0.0.0.0:8080->8080/tcp, :::8080->8080/tcp, 0.0.0.0:8090->8090/tcp, :::8090->8090/tcp   opendatacamnano_opendatacam_1
38c7bcddcadc   mongo                                 "docker-entrypoint.s…"   40 hours ago   Restarting (132) 5 seconds ago                                                                                                                                     opendatacamnano_mongo_1
```

kyutums, 2021-08-10:
I tried `sudo docker logs --tail 50 --follow --timestamps e234e724d863` (this is the new mongodb container as I reinstalled ODC to check if it would work; still the same problems) and it is still empty.

Also, every `sudo docker ps` I do shows the mongodb container restarting.

vsaw, 2021-08-10:
Try stopping both containers using `docker stop <container I'd>` and then try the following

Run `docker-compose up` in the folder with your docker-compose file in it. This should start both services and print the full Startup log in your terminal.

kyutums, 2021-08-10:
Thanks @vsaw . I tried the commands you noted. The results of the Startup log are below. It hasn't exited yet (still going after 5 mins.) though.

`Starting opendatacam_mongo_1       ... done
Starting opendatacam_opendatacam_1 ... done
Attaching to opendatacam_mongo_1, opendatacam_opendatacam_1
opendatacam_1  | mount: /tmp/tmpmount/console: special device /dev/console does not exist.
opendatacam_1  | umount: /dev: target is busy.
opendatacam_1  | Using video input value from config.json: "file"
opendatacam_1  | Using video input file value from config.json: "opendatacam_videos/demo.mp4"
opendatacam_1  | Using video input usbcam value from config.json: "v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink"
opendatacam_1  | Using video input remote_cam value from config.json: "YOUR IP CAM STREAM (can be .m3u8, MJPEG ...), anything supported by opencv"
opendatacam_mongo_1 exited with code 132
opendatacam_1  |
opendatacam_1  | > OpenDataCam@3.0.1 start /var/local/opendatacam
opendatacam_1  | > PORT=8080 NODE_ENV=production node server.js
opendatacam_1  |
opendatacam_mongo_1 exited with code 132
opendatacam_1  | -----------------------------------
opendatacam_1  | -     Opendatacam initialized     -
opendatacam_1  | - Config loaded:                  -
opendatacam_1  | {
opendatacam_1  |   "OPENDATACAM_VERSION": "3.0.1",
opendatacam_1  |   "PATH_TO_YOLO_DARKNET": "/var/local/darknet",
opendatacam_1  |   "VIDEO_INPUT": "file",
opendatacam_1  |   "NEURAL_NETWORK": "yolov4-tiny",
opendatacam_1  |   "VIDEO_INPUTS_PARAMS": {
opendatacam_1  |     "file": "opendatacam_videos/demo.mp4",
opendatacam_1  |     "usbcam": "v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink",
opendatacam_1  |     "raspberrycam": "nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink",
opendatacam_1  |     "remote_cam": "YOUR IP CAM STREAM (can be .m3u8, MJPEG ...), anything supported by opencv",
opendatacam_1  |     "remote_hls_gstreamer": "souphttpsrc location=http://YOUR_HLSSTREAM_URL_HERE.m3u8 ! hlsdemux ! decodebin ! videoconvert ! videoscale ! appsink"
opendatacam_1  |   },
opendatacam_1  |   "TRACKER_SETTINGS": {
opendatacam_1  |     "objectMaxAreaInPercentageOfFrame": 80,
opendatacam_1  |     "confidence_threshold": 0.2,
opendatacam_1  |     "iouLimit": 0.05,
opendatacam_1  |     "unMatchedFrameTolerance": 5
opendatacam_1  |   },
opendatacam_1  |   "COUNTER_SETTINGS": {
opendatacam_1  |     "minAngleWithCountingLineThreshold": 5,
opendatacam_1  |     "computeTrajectoryBasedOnNbOfPastFrame": 5
opendatacam_1  |   },
opendatacam_1  |   "VALID_CLASSES": [
opendatacam_1  |     "*"
opendatacam_1  |   ],
opendatacam_1  |   "DISPLAY_CLASSES": [
opendatacam_1  |     {
opendatacam_1  |       "class": "bicycle",
opendatacam_1  |       "hexcode": "1F6B2"
opendatacam_1  |     },
opendatacam_1  |     {
opendatacam_1  |       "class": "person",
opendatacam_1  |       "hexcode": "1F6B6"
opendatacam_1  |     },
opendatacam_1  |     {
opendatacam_1  |       "class": "truck",
opendatacam_1  |       "hexcode": "1F69B"
opendatacam_1  |     },
opendatacam_1  |     {
opendatacam_1  |       "class": "motorbike",
opendatacam_1  |       "hexcode": "1F6F5"
opendatacam_1  |     },
opendatacam_1  |     {
opendatacam_1  |       "class": "car",
opendatacam_1  |       "hexcode": "1F697"
opendatacam_1  |     },
opendatacam_1  |     {
opendatacam_1  |       "class": "bus",
opendatacam_1  |       "hexcode": "1F68C"
opendatacam_1  |     }
[... 63 lines cut ...]
opendatacam_1  |     at listOnTimeout (internal/timers.js:549:17)
opendatacam_1  |     at processTimers (internal/timers.js:492:7) {
opendatacam_1  |   reason: TopologyDescription {
opendatacam_1  |     type: 'Single',
opendatacam_1  |     setName: null,
opendatacam_1  |     maxSetVersion: null,
opendatacam_1  |     maxElectionId: null,
opendatacam_1  |     servers: Map { 'mongo:27017' => [ServerDescription] },
opendatacam_1  |     stale: false,
opendatacam_1  |     compatible: true,
opendatacam_1  |     compatibilityError: null,
opendatacam_1  |     logicalSessionTimeoutMinutes: null,
opendatacam_1  |     heartbeatFrequencyMS: 10000,
opendatacam_1  |     localThresholdMS: 15,
opendatacam_1  |     commonWireVersion: null
opendatacam_1  |   }
opendatacam_1  | }
opendatacam_mongo_1 exited with code 132
opendatacam_mongo_1 exited with code 132

opendatacam_mongo_1 exited with code 132
opendatacam_mongo_1 exited with code 132
opendatacam_mongo_1 exited with code 132
opendatacam_mongo_1 exited with code 132
opendatacam_mongo_1 exited with code 132`

kyutums, 2021-08-10:
I did as requested but I'm still experiencing a series of `opendatacam_mongo_1 exited with code 132`. I agree - it is totally odd. The closest similar problem [I found on the net](https://githubmemory.com/repo/PatrickHallek/automated-irrigation-system/issues/24) is about ARMv7, but I don't think it applies here.

I'll try doing a a fresh install of JetPack 4.3 for Nano on the SD card. I'm thinking that one of the depedencies got hit when I did a `sudo apt upgrade`.

vsaw, 2021-08-10:
You can also go for JetPack 4.5.1 as 4.3 is getting a bit old and this way we can make sure that dependencies are not eroding for ODC. 

We will update the dependency and docs for ODC to the new version of JetPack for the upcoming release anyway ;-)

kyutums, 2021-08-10:
Got it. I'll try it with JetPack 4.5.1. :)

kyutums, 2021-08-10:
I'm stumped. I did a complete reinstall from scratch. I used Jetpack 4.5.1 on the Nano. I used the instructions from https://github.com/opendatacam/opendatacam (but used the method in https://blog.hypriot.com/post/nvidia-jetson-nano-install-docker-compose/ to install docker-compose). But I still get the mongodb restarting issue. Results from `docker ps` is below.

> CONTAINER ID   IMAGE                                 COMMAND                  CREATED          STATUS                            PORTS                                                                                                                             NAMES
5b2747f7e131   opendatacam/opendatacam:v3.0.1-nano   "/usr/bin/entry.sh .…"   13 minutes ago   Up 2 minutes                      0.0.0.0:8070->8070/tcp, :::8070->8070/tcp, 0.0.0.0:8080->8080/tcp, :::8080->8080/tcp, 0.0.0.0:8090->8090/tcp, :::8090->8090/tcp   opendatacam_opendatacam_1
ed7ea1e4f065   mongo                                 "docker-entrypoint.s…"   13 minutes ago   Restarting (132) 39 seconds ago                                                                                                                                     opendatacam_mongo_1

vsaw, 2021-08-10:
Ah, try installing docker-compose using `apt-get` the packaged version now works out of the box (at least I never tried to build docker-compose myself and I never had any problems).

kyutums, 2021-08-10:
Additional information (in case it helps):

- The docker-compose install instructions from https://blog.hypriot.com/post/nvidia-jetson-nano-install-docker-compose/ didn't all work. I had to skip `curl -sSL https://bootstrap.pypa.io/get-pip.py | sudo python` because I got the following error: `ERROR: This script does not work on Python 2.7 The minimum supported Python version is 3.6. Please use https://bootstrap.pypa.io/pip/2.7/get-pip.py instead.` When I did a `pip --version`, I got pip `9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)` vs the expected `pip 19.0.3 from /usr/local/lib/python2.7/dist-packages/pip (python 2.7)`. Still, `sudo pip install docker-compose=="${DOCKER_COMPOSE_VERSION}"` worked after I typed it in.
- After the above, I tried `sudo pip3 install docker-compose` from the install instructions. It wasn't able to install due to missing cryptography so I had to install and updated version of rust then ```
sudo apt-get install build-essential libssl-dev libffi-dev \
    python3-dev cargo
``` before installing cryptography

kyutums, 2021-08-12:
Again another retry. Installed Jetpack 4.4.1 (no harm in trying via a lower version of Jetpack). Stumped again as the mongodb container is in a perpetual  `Restarting (132) About a minute ago` loop.  I am so unlucky with this. 🗡️ 
[image]

kyutums, 2021-08-12:
Could it possibly be a hardware problem?

vsaw, 2021-08-12:
I don't think so. Do you have by any chance a mongo instance running outside of docker? Or maybe something else that binds to port 27017?

What you could try is change the port in the docker-compose.yml and see if that helps. But I'm also at a loss right now. Unfortunately I don't have time to recreate the issue right now.

And just to cover all of our bases: Are you running it on the 2GB or 4GB version? And if it's the 2GB version, are you running a desktop environment on the Nano, or just SSH?

kyutums, 2021-08-13:
> I don't think so. Do you have by any chance a mongo instance running outside of docker? Or maybe something else that binds to port 27017?

As far as I know, the only running mongo is in Docker. I'll give it another check.

> What you could try is change the port in the docker-compose.yml and see if that helps. But I'm also at a loss right now. Unfortunately I don't have time to recreate the issue right now.

Oh! That's a good idea! I'll give it a whirl. :)

> And just to cover all of our bases: Are you running it on the 2GB or 4GB version? And if it's the 2GB version, are you running a desktop environment on the Nano, or just SSH?

I'm running it on the 4GB version. I've been installing it via SSH, but I'll try running it on the same device via its own browser

vsaw, 2021-08-13:
Okay, with the 4Gb Version you are fine. I was asking because we have seen people running out of RAM with the 2GB version if they have a Desktop environment running on the Nano as well. 

Good look with trying out different ports!

kyutums, 2021-08-13:
I tried changing the ports in the config.json, but the mongdb container was still stuck on Restarting. :(

Oh well. It was a valiant effort. :) Thank you @vsaw for all the help! It was quite a learning experience. :)

The journey continues as I try installing ODC on the Nano via the non Docker path. :)

closed 2021-08-13 by kyutums

vsaw, 2021-08-13:
Have you changed them in the docker-compose.yaml as well?

The config.Json only tells ODC where to find Mongo. It does not actually influence the port mongo will bind to

kyutums, 2021-08-14:
I _think_ I did. Looking back, I'm not so sure now. :D I'll try giving it a whirl again. :)

vsaw, 2021-08-14:
@kyutums could you take a look at This https://github.com/opendatacam/opendatacam/pull/426 And let me know if this works?

yauh, 2021-08-14:
Ran into the same issue. When I tried to install another instance of a docker container on my Synology NAS to point opendatacam to use it I saw the error message that the CPU was not supported for Mongo v5. Downgraded to 4.x and it worked flawlessly. Also when I did this on the Jetson, all was well. So indeed, it is a hardware support issue, mongo 5 introduces more requirements re: the CPU it is willing to run on. 🤷

vsaw, 2021-08-14:
Looks to me like a build error on Mongo side. Maybe the arch tag was set incorrectly for Mongo 5 as Docker usually complains if there is no image for target architecture. 🤷‍♂️

In any case great that you figured this out!

IssaIssa1, 2022-01-26:
I am having the same issue. I am running opendatacam on Jetson nano using Docker. When I start recording it shows that there's a file in the Data page but when I stop Data page turns blank. Any help would be appreciated