opendatacam/opendatacam #427: mongo containers crashes on Jetson Nano

yauh, 2021-08-14:
Beginning with Mongodb 5 specific cpu requirements prevent Jetson Nano from successfully running the mongo container from the compose file. Using docker-compose with an image tagged for a version 4 such as 
> image: mongo:4.4.8
will enable the mongo container to start successfully.

vsaw, 2021-08-14:
I've done some digging and you are correct. As of Mongo 5, a ARMv8.2 Microarch is required. 

This prevents to the Nano to run Mongo 5 as it only supports v8.0.

However, the Xavier NX does support v8.2 and should be able to run Mongo 5, so it really just affects the Nano for now. 

That being said I should double check if the Xavier NX can run Mongo 4.4 otherwise it will be really funny and we need to change the docker-compose based on target platform. 

Some links on the issue

- https://en.wikipedia.org/wiki/Comparison_of_ARMv8-A_cores
- https://developer.nvidia.com/embedded/jetson-nano-developer-kit
- https://developer.nvidia.com/blog/jetson-xavier-nx-the-worlds-smallest-ai-supercomputer/
- https://docs.mongodb.com/manual/administration/production-notes/#std-label-prod-notes-supported-platforms
- https://docs.mongodb.com/manual/release-notes/5.0/

vsaw, 2021-08-14:
I could confirm that Mongo 4.4.8 runs on Xavier NX. So your proposed solution is good to go!

vsaw, 2021-08-24:
Fix was merged into the release candidate v3.0.2. Will keep this issue open for now until the new release is out.

vsaw, 2021-09-16:
Closed as v3.0.2 is now released.

See also https://github.com/opendatacam/opendatacam/releases/tag/v3.0.2

closed 2021-09-16 by vsaw