opendatacam/opendatacam #428: Issues setting up multiple ODC instances on the same hardware using docker

twest49, 2021-08-17:
Hey, I am quite new here and making my first steps with ODC. I ran over some issues with using a second instance of ODC.

I've installed the desktop version (docker) like mentioned in the instructions. The first instance of ODC is running fine, but from the second instance I get the message 
> Didn't get video resolution yet, not sending tracker info

although I am using the same video source.

I copied the docker-compose.yml and the config.json in another folder. In the docker-compose.yml I changed the ports volumes and service names:
```
version: "2.3"
services:
  opendatacam_2:
    restart: always
    runtime: nvidia
    image: opendatacam/opendatacam:v3.0.1-desktop
    privileged: true
    volumes:
      - './config.json:/var/local/opendatacam/config.json'
    ports:
      - "8086:8080"
      - "8071:8070"
      - "8096:8090"
  mongo_2:
    image: mongo
    restart: always
    ports:
      - "27018:27017"
    volumes:
      - mongodb_data_container_2:/data/db
volumes:
  mongodb_data_container_2:
```
In the config.json the port of the mongodb is adapted, too. I started the second instance from this second folder with docker-compose up.
Is there some obvious mistake in my procedure?

closed 2021-08-18 by twest49

AlexHolly, 2021-09-07:
@twest49 Where you able to solve it? 

I have this issue:
When I change the name "opendatacam" in the docker-compose.yml to "opendatacam1" it keeps showing the loading screen after initializing the camera on http://localhost:8081/.

.env file (also tried to modify config.json)
COMPOSE_PROJECT_NAME=opendatacam1
PORT_APP=8081
PORT_DARKNET_MJPEG_STREAM=8090
PORT_DARKNET_JSON_STREAM=8070
MONGODB_URL=mongodb://mongo:27017

AlexHolly, 2021-09-07:
Found the issue https://github.com/opendatacam/opendatacam/blob/master/docker/build/desktop/launch.sh#L38

twest49, 2021-09-09:
My problem was fairly simple to solve. Both instances ran under different users, but in folders with the same name. ODC took the folders name as name for the container and the network-bridge. Hence renaming one folder solved the problem.