opendatacam/opendatacam #431: Crop webcam stream: Best approach?

JonasDaWi, 2021-08-25:
The area my webcam observes is quite huge. I only need the lower left quadrant to be evaluated. So my idea is to set the webcam to a high resolution, crop the videostream to that quadrant and let this be analyzed (400px*400px). What is the best approach to do so?
Also: is my thinking right that doing so should increase the accuracy?

Greetings!

vsaw, 2021-08-25:
Best if you crop the video by adjusting the gatreamer pipeline. Nvidia has some examples on how to do so. See e.g. https://developer.download.nvidia.com/embedded/L4T/r24_Release_v2.1/Docs/Accelerated_GStreamer_User_Guide_Release_24.2.1.pdf?kTboIMbiqc6kdfGqIuL_1mfRaPnerHKVXAvlWoAUfvokQR0zzIAc3GhltRPhdpnoZvuW-kM96I-AoIXJWCOnLxH8uuPi8dFaxC2j9G-dh4AYVQ1zRB9pVNmGrPhC_PqMQaBRZoiI2WGrbxZm-CUhTuCWqmBt8j-9iuXGAU3jamGYAYIkJWhACp0f8EGYVO2FUKjHn5g#page18

And yes this should also boost your detection.

zeljkomarinkovic, 2021-08-30:
will this work for `remote_cam` as well, since [remote_cam](https://github.com/opendatacam/opendatacam/blob/ae22eb8927aab5e8c116904587ae5ee728bb1bb0/config.json#L10) goes directly to opencv?

vsaw, 2021-08-30:
Yes, you just have to adjust the pipeline. 

Instead of just dumping the URL, you could e.g. use the RTSPSRC plugin directly and then format the stream before you pass it to Darknet.

vsaw, 2021-11-06:
Closed due to inactivity.

closed 2021-11-06 by vsaw