opendatacam/opendatacam #448: Clean up Darknet-Base Docker Tags

vsaw, 2021-09-16:
I've recently matched the darknet-base docker release process with ODC and as a result the following tags are obsolete and need to be removed

- docker pull opendatacam/opendatacam-darknet-base:xavier
- docker pull opendatacam/opendatacam-darknet-base:nano
- docker pull opendatacam/opendatacam-darknet-base:desktop
- docker pull opendatacam/opendatacam-darknet-base:xavier-development
- docker pull opendatacam/opendatacam-darknet-base:nano-development
- docker pull opendatacam/opendatacam-darknet-base:desktop-development

I'm assigning this to @tdurand as I don't have write access to dockerhub

tdurand, 2021-09-17:
Hi @vsaw  ! well done, sorry I'm off the internet those days, just sent you by email the dockerhub credentials

vsaw, 2021-09-17:
Done.

closed 2021-09-17 by vsaw