opendatacam/opendatacam #461: Docker container needs a restart when camera signal was lost for a short time

AlexHolly, 2021-10-01:
Version https://github.com/opendatacam/opendatacam/tree/db78e881097619616a5cb1e94b68abc3178c6798

When I unplug the camera for a short time (happens for both IP and USB) I have to restart the docker container otherwise the signal is never found again.

Anyone else has this issue or knows a solution?

vsaw, 2021-10-01:
It may be related to https://github.com/opendatacam/opendatacam/issues/368

Could you provide log data?

AlexHolly, 2021-10-01:
I also noticed the behavior described in #368 it also happens when I plug it in right after starting the container. I will try to provide a log tomorrow.

AlexHolly, 2021-10-02:
**Steps:**
1. Connect IP Camera
2. Start Docker Container
3. Disconnect IP Camera
4. Connect IP Camera

**Log:**
```
// Recognizing that stream is not available anymore is also not happening and I am not able to "Start Recording"

Start recording
Start recording
Start recording
Start recording
Start recording
Start recording
Start recording
Start recording
Start recording
Start recording
Start recording

// Here I connected the camera again

[NULL @ 0x564e1eac4500] missing picture in access unit

Stream closed.
input video stream closed. 
Close socket: out = 0, in = 0 
MJPG_sender: close clinet: 0 
Close socket: out = 0, in = 0 
MJPG_sender: close clinet: 0 
Close socket: out = 0, in = 0 
MJPG_sender: close clinet: 0 
Close socket: out = 0, in = 0 
MJPG_sender: close clinet: 0 
Close socket: out = 0, in = 0 
MJPG_sender: close acceptor: 0 

Close socket: out = 0, in = 0 
JSON_sender: close clinet: 0 
Close socket: out = 0, in = 0 
JSON_sender: close acceptor: 0 

==== HTTP Stream closed by darknet, reset UI ====
==== If you are running on a file, it is restarting  because you reached the end ====
==== If you are running on a camera, it might have crashed for some reason and we are trying to restart ====
Restart YOLO

// Many more CUDA lines

...

[yolo] params: iou loss: ciou (4), iou_norm: 0.07, cls_norm: 1.00, scale_x_y: 1.05
Total BFLOPS 60.137 
avg_outputs = 500162 
 Allocate additional workspace_size = 26.22 MB 
Demo
net.optimized_memory = 0 
mini_batch = 1, batch = 8, time_steps = 1, train = 0 
nms_kind: greedynms (1), beta = 0.600000 
nms_kind: greedynms (1), beta = 0.600000 
nms_kind: greedynms (1), beta = 0.600000 
Loading weights from yolov4.weights...Done! Loaded 162 layers from weights-file 
[tcp @ 0x55ca32fafa00] Connection to tcp://192.168.1.1:554?timeout=0 failed: No route to host

 seen 64, trained: 32032 K-images (500 Kilo-batches_64) 
video file: rtsp://secret:secret@192.168.1.1
[ WARN:0] global /var/local/git/opencv/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not open resource for reading and writing.
[ WARN:0] global /var/local/git/opencv/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /var/local/git/opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created

// Manually open http://localhost:8080/ in browser

already started
Something went wrong: connect ECONNREFUSED 172.19.0.4:8070
Too much retries, YOLO took more than 3 min to start, likely an error
-1
Something went wrong: connect ECONNREFUSED 172.19.0.4:8070
Too much retries, YOLO took more than 3 min to start, likely an error
-1

// Manually open http://localhost:8080/ in browser

already started
Something went wrong: connect ECONNREFUSED 172.19.0.4:8070
Too much retries, YOLO took more than 3 min to start, likely an error
-1
```

haviduck, 2021-10-09:
maybe due to 8080 already being yoinked at time of error and mjpeg being ad hoc?

vsaw, 2021-10-10:
Thank you @AlexHolly. That would indicate that Darknet is not starting up any more and would be a different error then #368. Last question, which Version of ODC are you running?

vsaw, 2021-10-21:
@AlexHolly Sorry I mussed have missed this then. This indicates that the Darknet process is not restarting. I can't really tell why without additional logs. Do you have any logs that show Darknet's or gstreamer output?

AlexHolly, 2021-11-03:
@vsaw How can I configure Darknet Docker to accept UDP. I can only see this PR but it was never merged https://github.com/AlexeyAB/darknet/pull/1976

vsaw, 2021-11-04:
As Darknet can take images from any Gstreamer pipeline we don't need to hack Darknet. Instead we will configure the GST pipeline.

The process has three steps.

# Step 1 Preparing ODC

I usually use something like this in my ODC configs

```
{
  "VIDEO_INPUT": "jpeg_remote_cam",
  "VIDEO_INPUTS_PARAMS": {
    "jpeg_remote_cam": "udpsrc port=7001 ! application/x-rtp,encoding-name=JPEG,payload=26 ! rtpjpegdepay ! jpegdec ! videoconvert ! videoscale ! appsink"
  },
  ...
}
```

# Step 2 - Configuring Port Forwarding

Now when you start the Docker container make sure that you forward UDP Port 7001 to ODC. You can skip this step if you don't use Docker.

See the [Docker Documentation on port Forwarding](https://docs.docker.com/engine/reference/commandline/run/#publish-or-expose-port--p---expose), but it should work a little something like this:

```
docker run -p 127.0.0.1:7001:7001/udp opendatacam/opendatacam:v3.0.2-nano
```

# Step 3 - Stream video via UDP

Now you should have ODC running in the background waiting for a UDP stream. You can stream your USB-Webcam to ODC when you run the following command in a different terminal

```
gst-launch-1.0 \
    v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! \
    nvjpegenc ! \
    rtpjpegpay ! \
    queue leaky=downstream max-size-buffers=1 ! \
    udpsink host=127.0.0.1 port=7001;
```

It may be the commands contain some typos or other errors but in general this is I've been using it. The great advantage is you can stop and start the command from Step 3 any time you want. From the point of ODC this is like having a frozen picture, but it does not crash ODC 😉

AlexHolly, 2021-11-30:
@vsaw Do you know how to make it run on a Desktop? The plugin `nvjpegenc` is missing?

https://forums.developer.nvidia.com/t/nvjpegenc-on-dgpu-platforms/84126

```
> gst-inspect-1.0 --version
gst-inspect-1.0 version 1.16.2
GStreamer 1.16.2
https://launchpad.net/distros/ubuntu/+source/gstreamer1.0
```

```
> gst-inspect-1.0 nvjpegdec
No such element or plugin 'nvjpegdec'
```

vsaw, 2021-11-30:
You can run it without the `nv` prefix. Will probably be slower frame rate but should work.

AlexHolly, 2021-11-30:
@vsaw It works like you described, but when I unplug the camera it will have a different video source (/dev/video0 or video1) and I have to rerun `gst-launch-1.0`. It is also not returning any error/signal to restart it automatically :(

```
gst-launch-1.0 \
    v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=480 ! \
    jpegenc ! \
    rtpjpegpay ! \
    queue leaky=downstream max-size-buffers=1 ! \
    udpsink host=127.0.0.1 port=7001;
```

---
My main use case is an Ip-Cam. Here I get at least a return value after a few seconds I will try tomorrow to automate the whole process and check if it is reliable:

```
gst-launch-1.0 \
    rtspsrc location=rtsp://admin:admin@192.168.0.100 ! \
    decodebin ! \
    videoconvert ! \
    jpegenc ! \
    rtpjpegpay ! \
    queue max-size-buffers=1 ! \
    udpsink host=127.0.0.1 port=7001;
```

Config:
```
udpsrc port=7001 ! application/x-rtp,encoding-name=JPEG,payload=26 ! rtpjpegdepay ! jpegdec ! videoconvert ! videoscale ! appsink
```

AlexHolly, 2021-12-01:
@vsaw I created a systemd service for the `gst-launch-1.0` looks good so far. One small issue. Now after "two minutes" of camera disconnect the ODC live preview shows a grey background and is not able to repair without a refresh (F5). It shows the detected areas but no image. It works when the camera is back before the "two minutes". Is there a way to change this?

Error:

Close socket: out = -1, in = -1 
[image]

vsaw, 2021-12-01:
This is a bug in MJPEG Proxy. It is just "cosmetic" the counting logic is not affected. 

We've discussed fixing it in https://github.com/opendatacam/opendatacam/pull/492
Feel free to give it a try if you like!

AlexHolly, 2021-12-01:
Ok, I followed the discussion on #492. If it is just cleaning up by adding 99999 retries without the config. I could give it a try this weekend.

Thank you for your help, feel free to close this issue.

vsaw, 2021-12-01:
If I remember correctly that's not enough. We discussed listening to the close event in the MJPEG Proxy. However I don't know what happens in your case with the UDP Stream config.