opendatacam/opendatacam #465: Development Branch:  Errors / Warnings on `NPM RUN START`

munsterlander, 2021-10-29:
When running `npm run start` for a non-docker build on the development branch, the following errors occur (although ODC does start successfully):

```
npm run start

> OpenDataCam@3.0.2 start /home/munsterlander/opendatacam
> NODE_ENV=production scripts/npm/startServer.sh

PORT=8080
NODE_ENV=production
{"message":"parser plugin 'param' not found in block: 1 in file: /home/mark/opendatacam/.build/apidoc/main.js","level":"warn"}
Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
  
   ⚠️  Tailwind is not purging unused styles because no template paths have been provided.
      If you have manually configured PurgeCSS outside of Tailwind or are deliberately not
      removing unused styles, set `purge: false` in your Tailwind config file to silence
      this warning.
   
      https://tailwindcss.com/docs/controlling-file-size/#removing-unused-css
Creating an optimized production build  

Compiled successfully.

Warning: You have opted-out of Automatic Static Optimization due to `getInitialProps` in `pages/_app`. This does not opt-out pages with `getStaticProps`
Read more: https://err.sh/next.js/opt-out-auto-static-optimization

Automatically optimizing pages  

Page                                                           Size     First Load JS
┌ λ /                                                          109 kB          205 kB
├   /_app                                                      3.21 kB        96.1 kB
└ λ /404                                                       2.55 kB        98.7 kB
+ First Load JS shared by all                                  96.1 kB
  ├ static/pages/_app.js                                       3.21 kB
  ├ chunks/2d1e5e5b.4c419f.js                                  64 B
  ├ chunks/7fd24589e5a0cc4e02d99df08bbc80c8870380b4.892f71.js  34.3 kB
  ├ chunks/affb85ec3cb120ccdb96e95b8df3f13afc33c6e7.ff32dd.js  8.19 kB
  ├ chunks/commons.b262b3.js                                   3.18 kB
  ├ chunks/framework.619a4f.js                                 40.3 kB
  ├ runtime/main.3a9b3f.js                                     6.23 kB
  ├ runtime/webpack.1c5199.js                                  746 B
  └ css/8436ea9a6681a2ae1b6b.css                               144 kB

λ  (Server)  server-side renders at runtime (uses getInitialProps or getServerSideProps)
○  (Static)  automatically rendered as static HTML (uses no initial props)
●  (SSG)     automatically generated as static HTML + JSON (uses getStaticProps)

-----------------------------------
-     Opendatacam initialized     -
- Config loaded:                  -
{
  "OPENDATACAM_VERSION": "3.0.2",
  "PATH_TO_YOLO_DARKNET": "/home/munsterlander/darknet",
  "CMD_TO_YOLO_DARKNET": "/home/munsterlander/darknet/darknet",
  "VIDEO_UPLOAD_FOLDER": "/home/munsterlander/darknet/opendatacam_videos_uploaded",
  "VIDEO_INPUT": "file",
  "NEURAL_NETWORK": "yolov4-tiny",
  "VIDEO_INPUTS_PARAMS": {
    "file": "opendatacam_videos_uploaded/demo.mp4",
    "usbcam": "v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink",
    "raspberrycam": "nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink",
    "remote_cam": "YOUR IP CAM STREAM (can be .m3u8, MJPEG ...), anything supported by opencv",
    "remote_hls_gstreamer": "souphttpsrc location=http://YOUR_HLSSTREAM_URL_HERE.m3u8 ! hlsdemux ! decodebin ! videoconvert ! videoscale ! appsink",
    "simulation": "--yolo_json public/static/placeholder/alexeydetections30FPS.json --video_file_or_folder public/static/placeholder/frames --isLive true --jsonFps 20 --mjpgFps 0.2"
  },
  "TRACKER_SETTINGS": {
    "objectMaxAreaInPercentageOfFrame": 80,
    "confidence_threshold": 0.2,
    "iouLimit": 0.05,
    "unMatchedFrameTolerance": 5,
    "fastDelete": true
  },
  "COUNTER_SETTINGS": {
    "countingAreaMinFramesInsideToBeCounted": 1,
    "countingAreaVerifyIfObjectEntersCrossingOneEdge": true,
    "minAngleWithCountingLineThreshold": 5,
    "computeTrajectoryBasedOnNbOfPastFrame": 5
  },
  "VALID_CLASSES": [
    "*"
  ],
  "DISPLAY_CLASSES": [
    {
      "class": "bicycle",
      "hexcode": "1F6B2"
    },
    {
      "class": "person",
      "hexcode": "1F6B6"
    },
    {
      "class": "truck",
      "hexcode": "1F69B"
    },
    {
      "class": "motorbike",
      "hexcode": "1F6F5"
    },
    {
      "class": "car",
      "hexcode": "1F697"
    },
    {
      "class": "bus",
      "hexcode": "1F68C"
    }
  ],
  "PATHFINDER_COLORS": [
    "#1f77b4",
    "#ff7f0e",
    "#2ca02c",
    "#d62728",
    "#9467bd",
    "#8c564b",
    "#e377c2",
    "#7f7f7f",
    "#bcbd22",
    "#17becf"
  ],
  "COUNTER_COLORS": {
    "yellow": "#FFE700",
    "turquoise": "#A3FFF4",
    "green": "#a0f17f",
    "purple": "#d070f0",
    "red": "#AB4435"
  },
  "NEURAL_NETWORK_PARAMS": {
    "yolov4": {
      "data": "cfg/coco.data",
      "cfg": "cfg/yolov4-416x416.cfg",
      "weights": "yolov4.weights"
    },
    "yolov4-tiny": {
      "data": "cfg/coco.data",
      "cfg": "cfg/yolov4-tiny.cfg",
      "weights": "yolov4-tiny.weights"
    }
  },
  "TRACKER_ACCURACY_DISPLAY": {
    "nbFrameBuffer": 300,
    "settings": {
      "radius": 3.1,
      "blur": 6.2,
      "step": 0.1,
      "gradient": {
        "1": "red",
        "0.4": "orange"
      },
      "canvasResolutionFactor": 0.1
    }
  },
  "DATABASE": "mongo",
  "DATABASE_PARAMS": {
    "mongo": {
      "url": "mongodb://127.0.0.1:27017"
    }
  },
  "PORTS": {
    "app": 8080,
    "darknet_json_stream": 8070,
    "darknet_mjpeg_stream": 8090
  },
  "GPS": {
    "enabled": false,
    "port": 2947,
    "hostname": "localhost",
    "signalLossTimeoutSeconds": 60,
    "csvExportOpenStreetMapsUrl": true
  }
}
-----------------------------------
Process YOLO initialized
> Ready on http://localhost:8080
> Ready on http://192.168.1.15:8080
Success init db
(node:10075) UnhandledPromiseRejectionWarning: MongoError: E11000 duplicate key error collection: opendatacam.recordings index: id_1 dup key: { : null }
    at MessageStream.messageHandler (/home/mark/opendatacam/node_modules/mongodb/lib/cmap/connection.js:261:20)
    at MessageStream.emit (events.js:314:20)
    at processIncomingData (/home/mark/opendatacam/node_modules/mongodb/lib/cmap/message_stream.js:144:12)
    at MessageStream._write (/home/mark/opendatacam/node_modules/mongodb/lib/cmap/message_stream.js:42:5)
    at doWrite (_stream_writable.js:403:12)
    at writeOrBuffer (_stream_writable.js:387:5)
    at MessageStream.Writable.write (_stream_writable.js:318:11)
    at Socket.ondata (_stream_readable.js:718:22)
    at Socket.emit (events.js:314:20)
    at addChunk (_stream_readable.js:297:12)
(node:10075) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:10075) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

```

vsaw, 2021-10-29:
This looks MongoDB related. Which MongoDB version are you running?

munsterlander, 2021-10-29:
Just the current default. I installed everything using the wiki for non-docker.  I'm not at my desk, but I will update with the version shortly.

vsaw, 2021-10-30:
Thanks. I've added it to the v4.0.0 release to make sure we don't forget about it

vsaw, 2022-02-13:
@munsterlander Did you run into this again? I have not yet been able to reproduce it and was wondering if we should keep it open for v4.0.0 or close it?

munsterlander, 2022-02-13:
No. It had to do with running mongo:latest and if that's been solved its good. I've been running dev branch on multiple builds without issue.

closed 2022-02-13 by munsterlander