opendatacam/opendatacam #467: Question:  Invoke OpenCV via Darknet (extend YoloDarknet.js??) OR install OpenCV via NPM as a package

munsterlander, 2021-11-03:
As the subject states, I need to access OpenCV functions but from within JS as I want to extend the classes to store lat/long of the detected objects and not of the receiver as mentioned in: https://github.com/opendatacam/opendatacam/pull/262.  In order to do this, I need specific OpenCV functions and I am not certain if its better to just include https://www.npmjs.com/package/opencv OR if its possible to extend `YoloDarknet.js` or something to have access to the underlying installation.  Any thoughts or recommendations on this?

vsaw, 2021-11-04:
So far we do not interface with OpenCV at all. All of this is taken care of by Darknet. Therefore I have no clue how to best interface with OpenCV. From your comment in https://github.com/opendatacam/opendatacam/pull/262#issuecomment-959586461 I'm wondering if you need OpenCV or if you can do the computation of the objects lat/lon in ODC, using the same Decorator patter as the GPS tracker.

I bring this up as the GPS tracker is actually quite small and easy to extend.

munsterlander, 2021-11-04:
@vsaw So we were playing with this and came to the conclusion that we are going to use a NodeJS <-> Python bridge, so that way we can call native calls to OpenCV.  Specifically, we need:  https://docs.opencv.org/2.4/modules/imgproc/doc/geometric_transformations.html#getperspectivetransform and I am sure we can write a pure JS implementation, but by going this route, we open up a lot more access to OpenCV's other functions that we can use.  I am going to use most of the GPS tracker, but then tweak it, so we will see what happens by the end of next week.  Thank you again for all the help and information!

closed 2021-11-04 by munsterlander