opendatacam/opendatacam #474: Issue with the unMatchedFrameTolerance setting

dennbas, 2021-11-05:
How to adjust the configuration in Opendatacam so that it tracks an object without interruption? 
In my case I am counting driving cars in a video. I noticed under pathfinder that the tracker loses the cars in the video and recognizes them as a new one a few frames later. (The line of the drawn path is interrupted and after a short pause it is drawn again with a new color). 

On [this page](https://opendatacam.github.io/opendatacam/documentation/CONFIG.html) was it recommend to increase "unMatchedFrameTolerance" from value. So I changed it from 5 to 20 and even higher values. Unfortunately, this did not bring any improvements, the error still occurs and does not change.

config.json (for upload as txt file):
[config.txt](https://github.com/opendatacam/opendatacam/files/7487460/config.txt)

Screenshot:
[image]

munsterlander, 2021-11-05:
Did you build after changing the config?

vsaw, 2021-11-06:
I also had the feeling that this sometimes does not work. But have never confirmed it yet. 

What I can offer as an alternative is changing the iouOverlap to a lower value. This could allow the tracker to keep an eye on faster moving cars. Have you tried tweaking this value as well?

parthapp, 2021-11-23:
I have had a similar issue as well. I will describe my project below. I am using Jetson Nano 2gb. 

I have a video recording from the corner of a traffic intersection with 4 approaches visible. I have labeled the approaches N, S, E and W for the detection lines. When a vehicle enters an intersection, it registers with a unique ID and timestamp. When it exits, it often registers with the same ID but with a different timestamp and a different line crossed. For a given length of video, I list all unique ID's, count the number of occurrences for each ID, and then list the Min and Max timestamp for each ID as well as the corresponding detection lines. This gives me a full count of all all vehicles entering an approach (N,S,E or W) exiting to a different departure direction. 

Ideally, number of unique ID's should be twice the vehicle count, and every ID should only show up twice. In my sample video, I had 20 double counts and 29 single counts. All single counts were vehicles that lost their initial ID and got assigned another on exit. I have increased unMatchedFrameTolerance and computeTrajectoryBasedOnNbOfPastFrame values alternatively and did a `sudo docker-compose restart` after every change, and compared against 20 double and 29 single counts that I get with default config.json. Those values dont change. I am either
1) adjusting the wrong parameters or
2) not letting ODC know that parameters are changed (incorrect "build"?)
3) parameters may not be working properly. 

I have not changed IOU settings, but I have seen tracker lose vehicles that were not occluded from beginning till end. 

Any suggestions?

vsaw, 2021-11-25:
Have you tried adjusting the iouLimit? It might be that the object is moving to fast for the tracker to follow its trajectory, a lower iouLimit could work with fast moving objects but might lead to false trajectories in crowded intersections as object may get merged.

Also I don't know if `sudo docker-compose restart` refreshes changes to the `config.json`  just to be safe I recommend fully stopping and starting ODC.

Last, which version are you running? Did you check if `development` behaves the same way?