opendatacam/opendatacam #476: Development: APIDocs warning

munsterlander, 2021-11-05:
When you do a build, you get this warning:

```
{"message":"parser plugin 'param' not found in block: 1 in file: /home/mark/ODC/opendatacam/.build/apidoc/main.js","level":"warn"}
```

This is because the output folder is included in the input folder while generating the api docs during the build process as documented here:  https://github.com/apidoc/apidoc/issues/672

We need to update: `/scripts/npm/generateapidoc.sh` https://github.com/munsterlander/opendatacam/blob/development/scripts/npm/generateapidoc.sh line 7, but I am uncertain of the downstream impact if I just change the paths.  Can anyone provide clarification on what it should be so we do not get this warning or are we ok with it?

vsaw, 2021-11-05:
Good Catch! Did you try what happens when we exclude .build from the APIdoc generation (like we do with node_modules)

munsterlander, 2021-11-06:
I didn't yet because I wasn't certain what the dependencies were.  I can give it a go in a bit.

munsterlander, 2021-11-06:
@vsaw So the code is in PR: https://github.com/opendatacam/opendatacam/pull/475 which excludes the directory and fixes the issue.  Additionally, with all the changes, now all the warnings are gone from the build process.  I added a comment at the bottom of that PR, but do you prefer one commit per PR or slam them all together like I did?  I wasn't really paying attention until the end there - apologies if that's now how you want it done.

vsaw, 2021-11-06:
Is like to keep this open until it is fixed in Development. If you don't mind.

munsterlander, 2021-11-06:
No worries at all.

vsaw, 2021-11-07:
Closed via #493

PR #493 (2021-11-07): 476 apidoc warning
Closes #476
commit 53b53b1d (2021-11-07): Chore: Fixes apidoc warning

Closes #476
closed 2021-11-07 by vsaw