opendatacam/opendatacam #477: Development:  Video Freezes as soon as you press start recording

munsterlander, 2021-11-05:
I know we have discussed this in other threads, but I am starting this thread to log the issue as we work on it in the hopes that someone sees something and goes, ah ha!  Anyway, here is the log when the freeze happens:

```
FPS:18.5         AVG_FPS:18.6
known client: 26, sent = 19309, must be sent outlen = 19309
problem with request:  Error: read ECONNRESET
    at TCP.onStreamRead (internal/stream_base_commons.js:209:20) {
  errno: 'ECONNRESET',
  code: 'ECONNRESET',
  syscall: 'read'
}
YOLO stream error: Error: read ECONNRESET
Too much retries, YOLO took more than 3 min to start, likely an error
Process YOLO exit
Process YOLO stopped
Process YOLO started
 CUDA-version: 10020 (10020), cuDNN: 8.2.1, GPU count: 1
 OpenCV version: 4.1.1
 0 : compute_capability = 530, cudnn_half = 0, GPU: NVIDIA Tegra X1
   layer   filters  size/strd(dil)      input                output
   0 conv     32       3 x 3/ 2    416 x 416 x   3 ->  208 x 208 x  32 0.075 BF
   1 conv     64       3 x 3/ 2    208 x 208 x  32 ->  104 x 104 x  64 0.399 BF
   2 conv     64       3 x 3/ 1    104 x 104 x  64 ->  104 x 104 x  64 0.797 BF
   3 route  2                                  1/2 ->  104 x 104 x  32
   4 conv     32       3 x 3/ 1    104 x 104 x  32 ->  104 x 104 x  32 0.199 BF
   5 conv     32       3 x 3/ 1    104 x 104 x  32 ->  104 x 104 x  32 0.199 BF
   6 route  5 4                                    ->  104 x 104 x  64
   7 conv     64       1 x 1/ 1    104 x 104 x  64 ->  104 x 104 x  64 0.089 BF
   8 route  2 7                                    ->  104 x 104 x 128
   9 max                2x 2/ 2    104 x 104 x 128 ->   52 x  52 x 128 0.001 BF
  10 conv    128       3 x 3/ 1     52 x  52 x 128 ->   52 x  52 x 128 0.797 BF
  11 route  10                                 1/2 ->   52 x  52 x  64
  12 conv     64       3 x 3/ 1     52 x  52 x  64 ->   52 x  52 x  64 0.199 BF
  13 conv     64       3 x 3/ 1     52 x  52 x  64 ->   52 x  52 x  64 0.199 BF
  14 route  13 12                                  ->   52 x  52 x 128
  15 conv    128       1 x 1/ 1     52 x  52 x 128 ->   52 x  52 x 128 0.089 BF
  16 route  10 15                                  ->   52 x  52 x 256
  17 max                2x 2/ 2     52 x  52 x 256 ->   26 x  26 x 256 0.001 BF
  18 conv    256       3 x 3/ 1     26 x  26 x 256 ->   26 x  26 x 256 0.797 BF
  19 route  18                                 1/2 ->   26 x  26 x 128
  20 conv    128       3 x 3/ 1     26 x  26 x 128 ->   26 x  26 x 128 0.199 BF
  21 conv    128       3 x 3/ 1     26 x  26 x 128 ->   26 x  26 x 128 0.199 BF
  22 route  21 20                                  ->   26 x  26 x 256
  23 conv    256       1 x 1/ 1     26 x  26 x 256 ->   26 x  26 x 256 0.089 BF
  24 route  18 23                                  ->   26 x  26 x 512
  25 max                2x 2/ 2     26 x  26 x 512 ->   13 x  13 x 512 0.000 BF
  26 conv    512       3 x 3/ 1     13 x  13 x 512 ->   13 x  13 x 512 0.797 BF
  27 conv    256       1 x 1/ 1     13 x  13 x 512 ->   13 x  13 x 256 0.044 BF
  28 conv    512       3 x 3/ 1     13 x  13 x 256 ->   13 x  13 x 512 0.399 BF
  29 conv    255       1 x 1/ 1     13 x  13 x 512 ->   13 x  13 x 255 0.044 BF
  30 yolo
[yolo] params: iou loss: ciou (4), iou_norm: 0.07, obj_norm: 1.00, cls_norm: 1.00, delta_norm: 1.00, scale_x_y: 1.05
  31 route  27                                     ->   13 x  13 x 256
  32 conv    128       1 x 1/ 1     13 x  13 x 256 ->   13 x  13 x 128 0.011 BF
  33 upsample                 2x    13 x  13 x 128 ->   26 x  26 x 128
  34 route  33 23                                  ->   26 x  26 x 384
  35 conv    256       3 x 3/ 1     26 x  26 x 384 ->   26 x  26 x 256 1.196 BF
  36 conv    255       1 x 1/ 1     26 x  26 x 256 ->   26 x  26 x 255 0.088 BF
  37 yolo
[yolo] params: iou loss: ciou (4), iou_norm: 0.07, obj_norm: 1.00, cls_norm: 1.00, delta_norm: 1.00, scale_x_y: 1.05
Total BFLOPS 6.910
avg_outputs = 310203
 Allocate additional workspace_size = 26.22 MB
Loading weights from yolov4-tiny.weights...Demo
net.optimized_memory = 0
mini_batch = 1, batch = 1, time_steps = 1, train = 0
Create CUDA-stream - 0
 Create cudnn-handle 0
nms_kind: greedynms (1), beta = 0.600000
nms_kind: greedynms (1), beta = 0.600000
Done! Loaded 38 layers from weights-file

```

vsaw, 2021-11-06:
Would it be possible to see the `config.JSON` and the complete log from the beginning until you get the error?

Also do you run ODC from a live feed or a video file?

munsterlander, 2021-11-06:
We are just running the demo video for now.  Here is the config file:

```
{
  "OPENDATACAM_VERSION": "3.0.2",
  "PATH_TO_YOLO_DARKNET" : "/home/mark/ODC/darknet",
  "CMD_TO_YOLO_DARKNET" : "/home/mark/ODC/darknet/darknet",
  "VIDEO_UPLOAD_FOLDER": "/home/mark/ODC/darknet/opendatacam_videos_uploaded",
  "VIDEO_INPUT": "file",
  "NEURAL_NETWORK": "yolov4-tiny",
  "VIDEO_INPUTS_PARAMS": {
    "file": "opendatacam_videos_uploaded/demo.mp4",
    "usbcam": "v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink",
    "raspberrycam": "nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink",
    "remote_cam": "YOUR IP CAM STREAM (can be .m3u8, MJPEG ...), anything supported by opencv",
    "remote_hls_gstreamer": "souphttpsrc location=http://YOUR_HLSSTREAM_URL_HERE.m3u8 ! hlsdemux ! decodebin ! videoconvert ! videoscale ! appsink",
    "simulation": "--yolo_json public/static/placeholder/alexeydetections30FPS.json --video_file_or_folder public/static/placeholder/frames --isLive true --jsonFps 20 --mjpgFps 0.2"
  },
  "TRACKER_SETTINGS": {
    "objectMaxAreaInPercentageOfFrame": 80,
    "confidence_threshold": 0.2,
    "iouLimit": 0.05,
    "unMatchedFrameTolerance": 5,
    "fastDelete": true
  },
  "COUNTER_SETTINGS": {
    "countingAreaMinFramesInsideToBeCounted": 1,
    "countingAreaVerifyIfObjectEntersCrossingOneEdge": true,
    "minAngleWithCountingLineThreshold": 5,
    "computeTrajectoryBasedOnNbOfPastFrame": 5
  },
  "VALID_CLASSES": ["*"],
  "DISPLAY_CLASSES": [
    { "class": "bicycle", "hexcode": "1F6B2"},
    { "class": "person", "hexcode": "1F6B6"},
    { "class": "truck", "hexcode": "1F69B"},
    { "class": "motorbike", "hexcode": "1F6F5"},
    { "class": "car", "hexcode": "1F697"},
    { "class": "bus", "hexcode": "1F68C"}
  ],
  "PATHFINDER_COLORS": [
    "#1f77b4",
    "#ff7f0e",
    "#2ca02c",
    "#d62728",
    "#9467bd",
    "#8c564b",
    "#e377c2",
    "#7f7f7f",
    "#bcbd22",
    "#17becf"
  ],
  "COUNTER_COLORS": {
    "yellow": "#FFE700",
    "turquoise": "#A3FFF4",
    "green": "#a0f17f",
    "purple": "#d070f0",
    "red": "#AB4435"
  },
  "NEURAL_NETWORK_PARAMS": {
    "yolov4": {
      "data": "cfg/coco.data",
      "cfg": "cfg/yolov4-416x416.cfg",
      "weights": "yolov4.weights"
    },
    "yolov4-tiny": {
      "data": "cfg/coco.data",
      "cfg": "cfg/yolov4-tiny.cfg",
      "weights": "yolov4-tiny.weights"
    }
[... 1231 lines cut ...]

FPS:18.3 	 AVG_FPS:18.5
 MJPEG-stream sent. 
Objects:

 JSON-stream sent. 

FPS:18.4 	 AVG_FPS:18.5
 MJPEG-stream sent. 
Objects:

 JSON-stream sent. 

FPS:18.4 	 AVG_FPS:18.5
 MJPEG-stream sent. 
Objects:

 JSON-stream sent. 

FPS:18.4 	 AVG_FPS:18.5
 MJPEG-stream sent. 
Objects:

 JSON-stream sent. 
```

vsaw, 2021-11-06:
Aha! Now I know what's going on!

It's just another variation of https://github.com/opendatacam/opendatacam/issues/368 you're seeing!

When running from File, ODC restarts Darknet to process the file from the beginning when you hit record. After restarting Darknet, ODC makes a few attempts to connect to Darknet. If that does not work it will not try again and not show any video or tracker data. 

So to fix this I suggest you look at Opendatacam.js and look for the Darknet connection in the `listenToYolo` function!

closed 2021-11-06 by vsaw

munsterlander, 2021-11-06:
Excellent!  Ok, so this is happening even in headless mode but we will dive into that thread!