opendatacam/opendatacam #495: MONGODB issue?

parthapp, 2021-11-09:
Hello, I am new to linux and opendatacam. I followed the install instructions for Jecton Nano 2gb and install failed. Something to do with Python2.7. I googled and was able to install opendatacam. It starts but cannot log data. I keep seeing this error repeat in the console. I cannot download the csv files, Chrome says its a network error. 

(node:190) UnhandledPromiseRejectionWarning: MongoServerSelectionError: getaddrinfo ENOTFOUND mongo
    at Timeout._onTimeout (/var/local/opendatacam/node_modules/mongodb/lib/core/sdam/topology.js:430:30)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
(node:190) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1653)

munsterlander, 2021-11-09:
Hello, are you running in Docker or out of Docker?  Can you also paste your config.json file?  In short, it is unable to locate your MongoDB service.  Can you verify it is running by `systemctl status mongod`?

parthapp, 2021-11-09:
I am running opendatacam inside docker, as per the tutorial for Jetson nano.  sudo docker-compose restart says both opendatacam and mongo_1 are "done". 
systemctl status mongod says "Unit mongod.service could not be found"

parthapp, 2021-11-09:
Complete list of steps:
1. Downloaded and flashed [https://developer.nvidia.com/jetson-nano-2gb-sd-card-image]image
2. Followed instructions [https://github.com/opendatacam/opendatacam/blob/master/documentation/jetson/FLASH_JETSON.md#Jetson-Nano]here and did the following
a) `sudo apt update`
b) `sudo apt install python-pip`
c) `sudo -H pip install jetson-stats`
d)` jetson_release`
e) `sudo apt-get install nano`
f) `sudo nano .bashrc` and added below lines to bottom of file
export PATH=${PATH}:/usr/local/cuda/bin
export LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:/usr/local/cuda/lib64

g) Install swap by:
```
git clone https://github.com/JetsonHacksNano/installSwapfile
cd installSwapfile
chmod 777 installSwapfile.sh
./installSwapfile.sh
sudo reboot
```

```
sudo apt install python3-pip

sudo apt-get install -y libffi-dev
sudo apt-get install -y python-openssl
sudo apt-get install libssl-dev
sudo pip3 install docker-compose
```
I get errors: 

The directory '/home/partha/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/partha/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.

And "Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-ibg_37zy/cryptography"

Proceed with install steps:
```wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/v3.0.1/docker/install-opendatacam.sh
chmod 777 install-opendatacam.sh
./install-opendatacam.sh --platform nano
```

I get error  "sudo: docker-compose: command not found"

I do this:
`python3 -m pip install --upgrade pip `
and then install docker compose again. Still warnings,  "WARNING: The directory '/home/partha/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you should use sudo's -H flag." but installs. Next try to install ODC again: It installs.  Then `sudo reboot`

I am able to start ODC at this point but then I start seeing the errors I posted initially. 

I included all the steps because I realize that packages used at the time of tutorial creation were different from what I am downloading today. FYI, I have done these steps with the Jetpack 4.3 image as well and same issue. 

I hoped that deploy with Balena would be a simpler process, but Balena image for Jetson does not show up on Balena dashboard although I can see Nano listed online with IP on my router.

munsterlander, 2021-11-10:
Try `sudo pip3 -H install docker-compose`

parthapp, 2021-11-10:
Hello Sir

Thanks for the response. I get this error when I try it.

***@***.***:~$ sudo pip3 -H install docker-compose
[sudo] password for partha:
WARNING: pip is being invoked by an old script wrapper. This will fail in a
future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the
underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of
running pip directly.

Usage:
  pip3 <command> [options]

no such option: -H

parthapp, 2021-11-11:
Console log file attached. The /var/local in the console log does not have any subdirectories or files. 

[console log.txt](https://github.com/opendatacam/opendatacam/files/7518101/console.log.txt)
[image]

munsterlander, 2021-11-11:
So this would mean that mongo is not started or is not at the address provided in the config.json file. I assume the previous command I gave still results in service not found. If so, you need to get mongo installed or started.

munsterlander, 2021-11-11:
Also, sounds like the docker instance isn't running correctly.

parthapp, 2021-11-11:
sudo docker-compose up
Creating network "partha_default" with the default driver
Creating partha_mongo_1       ... done
Creating partha_opendatacam_1 ... done
Attaching to partha_opendatacam_1, partha_mongo_1
opendatacam_1  | mount: /tmp/tmpmount/console: special device /dev/console
does not exist.
opendatacam_1  | umount: /dev: target is busy.
opendatacam_1  | Using video input value from config.json: "file"
opendatacam_1  | Using video input file value from config.json:
"opendatacam_videos/demo.mp4"
opendatacam_1  | Using video input usbcam value from config.json: "v4l2src
device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 !
videoconvert ! appsink"
opendatacam_1  | Using video input remote_cam value from config.json: "YOUR
IP CAM STREAM (can be .m3u8, MJPEG ...), anything supported by opencv"
partha_mongo_1 exited with code 132
partha_mongo_1 exited with code 132
opendatacam_1  |
opendatacam_1  | > ***@***.*** start /var/local/opendatacam
opendatacam_1  | > PORT=8080 NODE_ENV=production node server.js
opendatacam_1  |
partha_mongo_1 exited with code 132
opendatacam_1  | -----------------------------------
opendatacam_1  | -     Opendatacam initialized     -
opendatacam_1  | - Config loaded:                  -
opendatacam_1  | {
opendatacam_1  |   "OPENDATACAM_VERSION": "3.0.1",
opendatacam_1  |   "PATH_TO_YOLO_DARKNET": "/var/local/darknet",
opendatacam_1  |   "VIDEO_INPUT": "file",
opendatacam_1  |   "NEURAL_NETWORK": "yolov4-tiny",
opendatacam_1  |   "VIDEO_INPUTS_PARAMS": {
opendatacam_1  |     "file": "opendatacam_videos/demo.mp4",
opendatacam_1  |     "usbcam": "v4l2src device=/dev/video0 ! video/x-raw,
framerate=30/1, width=640, height=360 ! videoconvert ! appsink",
opendatacam_1  |     "raspberrycam": "nvarguscamerasrc !
video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1,
format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 !
videoconvert ! video/x-raw, format=BGR ! appsink",
opendatacam_1  |     "remote_cam": "YOUR IP CAM STREAM (can be .m3u8, MJPEG
...), anything supported by opencv",
opendatacam_1  |     "remote_hls_gstreamer": "souphttpsrc location=
http://YOUR_HLSSTREAM_URL_HERE.m3u8 ! hlsdemux ! decodebin ! videoconvert !
videoscale ! appsink"
opendatacam_1  |   },
opendatacam_1  |   "TRACKER_SETTINGS": {
opendatacam_1  |     "objectMaxAreaInPercentageOfFrame": 80,
opendatacam_1  |     "confidence_threshold": 0.2,
opendatacam_1  |     "iouLimit": 0.05,
opendatacam_1  |     "unMatchedFrameTolerance": 5
opendatacam_1  |   },
opendatacam_1  |   "COUNTER_SETTINGS": {
opendatacam_1  |     "minAngleWithCountingLineThreshold": 5,
opendatacam_1  |     "computeTrajectoryBasedOnNbOfPastFrame": 5
opendatacam_1  |   },
opendatacam_1  |   "VALID_CLASSES": [
opendatacam_1  |     "*"
opendatacam_1  |   ],
opendatacam_1  |   "DISPLAY_CLASSES": [
opendatacam_1  |     {
opendatacam_1  |       "class": "bicycle",
opendatacam_1  |       "hexcode": "1F6B2"
opendatacam_1  |     },
opendatacam_1  |     {
opendatacam_1  |       "class": "person",
opendatacam_1  |       "hexcode": "1F6B6"
opendatacam_1  |     },
opendatacam_1  |     {
opendatacam_1  |       "class": "truck",
opendatacam_1  |       "hexcode": "1F69B"
[... 73 lines cut ...]
partha_mongo_1 exited with code 132
partha_mongo_1 exited with code 132
opendatacam_1  | MongoServerSelectionError: getaddrinfo ENOTFOUND mongo
opendatacam_1  |     at Timeout._onTimeout
(/var/local/opendatacam/node_modules/mongodb/lib/core/sdam/topology.js:430:30)
opendatacam_1  |     at listOnTimeout (internal/timers.js:549:17)
opendatacam_1  |     at processTimers (internal/timers.js:492:7) {
opendatacam_1  |   reason: TopologyDescription {
opendatacam_1  |     type: 'Single',
opendatacam_1  |     setName: null,
opendatacam_1  |     maxSetVersion: null,
opendatacam_1  |     maxElectionId: null,
opendatacam_1  |     servers: Map { 'mongo:27017' => [ServerDescription] },
opendatacam_1  |     stale: false,
opendatacam_1  |     compatible: true,
opendatacam_1  |     compatibilityError: null,
opendatacam_1  |     logicalSessionTimeoutMinutes: null,
opendatacam_1  |     heartbeatFrequencyMS: 10000,
opendatacam_1  |     localThresholdMS: 15,
opendatacam_1  |     commonWireVersion: null
opendatacam_1  |   }
opendatacam_1  | }
partha_mongo_1 exited with code 132
partha_mongo_1 exited with code 132
partha_mongo_1 exited with code 132

munsterlander, 2021-11-11:
@parthapp Can you close one of these issues are they are duplicates?

parthapp, 2021-11-11:
Yes sir, done

munsterlander, 2021-11-11:
If I were you, I would wipe the install and just follow the instructions here: https://github.com/opendatacam/opendatacam#2-install-and-start-opendatacam- It downloads a script that installs everything for you. Also as previously mentioned, you should run sudo apt-get install docker-compose to install docker-compose and that should be all you need. I have tested and verified this works on nano's using Jetpack 4.3, 4.5, and 4.6. It will probably work on 4.4, I just never ran it on that.

munsterlander, 2021-11-11:
@vsaw They are reporting that the Dockerfile for the nano that is using the Balena image is installing Mongo 5 and not 4.  I have confirmed on a separate build that Mongo 5 does not start even though they say it will run on ARM64 - just not the Jetson.  I don't have a way to verify the Docker build right now as both of my Jetsons are deep into the work I am currently doing.

parthapp, 2021-11-12:
Hello All, not sure if its relevant, but this is what I was trying to deploy using Balena, but none of the Balena images I have tried boot and report as online on the dashboard. 
[image]

munsterlander, 2021-11-12:
Well that is Mongo 4.2 so that should work fine.  I don't know as I don't use Balena at all.  I just have done clean installs of the Jetson 4.6 image, then follow the instructions exactly after I SSH into the Jetson and it has always worked.

parthapp, 2021-11-12:
Thats what it was also doing till 11/05 and it installed and worked without
issue. Things changed when I tried again on 11/06 evening.  It has not
worked at all since that day.

parthapp, 2021-11-12:
Folks, I am happy to report a finding and an inelegant solution to the issue. When I changed mongo to mongo:4.2.14, ODC started working normally again. Full steps below, specifically for Jetson Nano 2gb:

Follow steps in the [guide](https://github.com/opendatacam/opendatacam) exactly for step 1. 

For step 2, do the following:

Download the script for ODC install.
wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/v3.0.1/docker/install-opendatacam.sh

Create a docker-compose.yml file, specific to the Nano and paste the things below. 
 
`sudo nano docker-compose.yml`
```
version: "2.3"
services:
  opendatacam:
    restart: always
    image: opendatacam/opendatacam:v3.0.1-nano
    privileged: true
    volumes:
      - './config.json:/var/local/opendatacam/config.json'
    ports:
      - "8080:8080"
      - "8070:8070"
      - "8090:8090"
  mongo:
    image: mongo:4.2.14
    restart: always
    ports:
      - "27017:27017"
    volumes:
      - mongodb_data_container:/data/db
volumes:
  mongodb_data_container:
```
Then, crtl X, Y and enter to save and exit. 
Now we edit out a line in the previously downloaded install-opendatacam.sh that downloads docker-compose.yml (without mongo version specified)  by adding a # to the begining of the line, as shown below. 
sudo nano install-opendatacam.sh
`  #wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/$VERSION/docker/run/$PLATFORM/docker-compose.yml`

then we make install-opendatacam.sh executable by typing (step 2 under the "Install commands)
`chmod 777 install-opendatacam.sh`

Step 3 of "Install commands"
./install-opendatacam.sh --platform nano

munsterlander, 2021-11-12:
Excellent!

parthapp, 2021-11-12:
I found an inelegant solution to the issue. I can confirm that it was mondo:latest that was the cause. Solution was to edit docker-compose.yml and specify mongo:4.2.14. Complete steps below for the next newcomer to linux and ODC :)

> Excellent!

Thank you sir! It was your finding that Mongo 5 does not work with Jetson specifically that was the key. Maybe this solution applies to other Jetsons as well.

closed 2021-11-12 by parthapp

munsterlander, 2021-11-12:
Will do.

parthapp, 2021-11-12:
Awesome, thanks!

vsaw, 2021-11-12:
Sorry guys this is my mistake I accidentally reverted master to v3.0.1. 

All of this is already fixed since 2 months in v3.0.2

Please select the release manually via the git tag until I reverted the branch (not sure I can do this from my phone and I don't know when I will have access to my machine again)