opendatacam/opendatacam #496: Updated tutorial for ODC installation on a Jetson Nano 2GB

parthapp, 2021-11-11:
Hello All, I am new to ODC and Linux generally and I am unable to install ODC following the guide on the page dedicated to the Jetson Nano, [here](https://github.com/opendatacam/opendatacam/blob/master/documentation/jetson/JETSON_NANO.md). 

Issue is this: In following the steps exactly, there are errors and warnings and Docker Compose does not install. Googling ways to make it work, I got it installed and ODC running, but I cannot get any CSV or JSON data out of it and the console lists errors continuously. My noob understanding is that ODC and mongodb are not talking to each other, a consequence of copy paste sudo commands from google without any understanding of them. I am using a Jetson Nano 2GB. 

I am requesting that the guide be updated. Thanks a lot and I really appreciate the effort put into this software!

vsaw, 2021-11-11:
Hey, few remarks here 

1. Can you please provide any logs? Ideally you can provide the docker logs and ODC logs. 
2. The docker-compose install is outdated. Just try the following `sudo apt-get install docker-compose`

Lastly, you're welcome to ask for help at any time, but please remember that this is an Open Source project we maintain in our free time. We ask every member of the community to please respect that, and I believe that "requesting" things is not the right tone for that. Thank you for understanding.

parthapp, 2021-11-11:
Hello Sir

Thanks for the response.
How do I access ODC logs?

Docker-compose log is below:
sudo docker-compose logs
Attaching to partha_mongo_1, partha_opendatacam_1
opendatacam_1  | mount: /tmp/tmpmount/console: special device /dev/console
does not exist.
opendatacam_1  | umount: /dev: target is busy.
opendatacam_1  | Using video input value from config.json: "file"
opendatacam_1  | Using video input file value from config.json:
"opendatacam_videos/demo.mp4"
opendatacam_1  | Using video input usbcam value from config.json: "v4l2src
device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 !
videoconvert ! appsink"
opendatacam_1  | Using video input remote_cam value from config.json: "YOUR
IP CAM STREAM (can be .m3u8, MJPEG ...), anything supported by opencv"
opendatacam_1  |
opendatacam_1  | > ***@***.*** start /var/local/opendatacam
opendatacam_1  | > PORT=8080 NODE_ENV=production node server.js
opendatacam_1  |
opendatacam_1  | -----------------------------------
opendatacam_1  | -     Opendatacam initialized     -
opendatacam_1  | - Config loaded:                  -
opendatacam_1  | {
opendatacam_1  |   "OPENDATACAM_VERSION": "3.0.1",
opendatacam_1  |   "PATH_TO_YOLO_DARKNET": "/var/local/darknet",
opendatacam_1  |   "VIDEO_INPUT": "file",
opendatacam_1  |   "NEURAL_NETWORK": "yolov4-tiny",
opendatacam_1  |   "VIDEO_INPUTS_PARAMS": {
opendatacam_1  |     "file": "opendatacam_videos/demo.mp4",
opendatacam_1  |     "usbcam": "v4l2src device=/dev/video0 ! video/x-raw,
framerate=30/1, width=640, height=360 ! videoconvert ! appsink",
opendatacam_1  |     "raspberrycam": "nvarguscamerasrc !
video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1,
format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 !
videoconvert ! video/x-raw, format=BGR ! appsink",
opendatacam_1  |     "remote_cam": "YOUR IP CAM STREAM (can be .m3u8, MJPEG
...), anything supported by opencv",
opendatacam_1  |     "remote_hls_gstreamer": "souphttpsrc location=
http://YOUR_HLSSTREAM_URL_HERE.m3u8 ! hlsdemux ! decodebin ! videoconvert !
videoscale ! appsink"
opendatacam_1  |   },
opendatacam_1  |   "TRACKER_SETTINGS": {
opendatacam_1  |     "objectMaxAreaInPercentageOfFrame": 80,
opendatacam_1  |     "confidence_threshold": 0.2,
opendatacam_1  |     "iouLimit": 0.05,
opendatacam_1  |     "unMatchedFrameTolerance": 5
opendatacam_1  |   },
opendatacam_1  |   "COUNTER_SETTINGS": {
opendatacam_1  |     "minAngleWithCountingLineThreshold": 5,
opendatacam_1  |     "computeTrajectoryBasedOnNbOfPastFrame": 5
opendatacam_1  |   },
opendatacam_1  |   "VALID_CLASSES": [
opendatacam_1  |     "*"
opendatacam_1  |   ],
opendatacam_1  |   "DISPLAY_CLASSES": [
opendatacam_1  |     {
opendatacam_1  |       "class": "bicycle",
opendatacam_1  |       "hexcode": "1F6B2"
opendatacam_1  |     },
opendatacam_1  |     {
opendatacam_1  |       "class": "person",
opendatacam_1  |       "hexcode": "1F6B6"
opendatacam_1  |     },
opendatacam_1  |     {
opendatacam_1  |       "class": "truck",
opendatacam_1  |       "hexcode": "1F69B"
[... 467 lines cut ...]
layers from weights-file
opendatacam_1  | Got first message from JSONStream
opendatacam_1  | Error with message send by YOLO, not valid JSON
opendatacam_1  | setvideoresolution
opendatacam_1  |
opendatacam_1  |  seen 64, trained: 32012 K-images (500 Kilo-batches_64)
opendatacam_1  | video file: opendatacam_videos/demo.mp4
opendatacam_1  | Video stream: 640 x 360
opendatacam_1  | JSON_sender: new client 27
opendatacam_1  | Start recording
opendatacam_1  | MJPG_sender: new client 29
opendatacam_1  | Stop recording
opendatacam_1  | (node:142) UnhandledPromiseRejectionWarning:
MongoServerSelectionError: getaddrinfo ENOTFOUND mongo
opendatacam_1  |     at Timeout._onTimeout
(/var/local/opendatacam/node_modules/mongodb/lib/core/sdam/topology.js:430:30)
opendatacam_1  |     at listOnTimeout (internal/timers.js:549:17)
opendatacam_1  |     at processTimers (internal/timers.js:492:7)
opendatacam_1  | (node:142) UnhandledPromiseRejectionWarning: Unhandled
promise rejection. This error originated either by throwing inside of an
async function without a catch block, or by rejecting a promise which was
not handled with .catch(). To terminate the node process on unhandled
promise rejection, use the CLI flag `--unhandled-rejections=strict` (see
https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection
id: 3)

munsterlander, 2021-11-11:
If I were you, I would wipe the install and just follow the instructions here:  https://github.com/opendatacam/opendatacam#2-install-and-start-opendatacam-  It downloads a script that installs everything for you.

closed 2021-11-11 by parthapp

parthapp, 2021-11-11:
Hello Sir. That is the link I started with and it was working perfectly
till 11/05. Since evening 11/06, that script has not worked for me. An
observation: the balena version of ODC uses mongodb 4.2.x and I noticed
that my docker had mongo "latest" installed. I have tried and failed to
install an older mongodb. I will try the script again and confirm if it
works now with 4.3.
Thanks again for your help. Really appreciate it.

vsaw, 2021-11-12:
This was because I accidentally reverted the master branch to v3.0.1 which contains outdated installation instructions.

I will revert master back to v3.0.2 when I get the chance. In the mean time please download and view v3.0.2 manually via the GitHub release page.