opendatacam/opendatacam #500: Question:  usbcam in config fails for one camera but not the other

munsterlander, 2021-11-21:
So this is more of a config question.  Using the default usbcam settings in `config.json`, it works perfectly with a usb webcam.  I am trying to use a wireless FPV camera setup and it fails.  Here is the output:

```
v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.200s (5.000 fps)

```

And here is the config:

```
"usbcam": "v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink",
```

Honestly, I don't have enough knowledge about these devices to add more information, so I am hoping someone knows a setting I am missing or something.  Lastly, what happens in ODC is when I am running the FPV cam, it just shows Video Stopped in the terminal when the other cam works perfectly (only 1 cam is plugged in at a time.)

vsaw, 2021-11-25:
I don't think it has anything to do with the MJPEG format. To test this you could just find any MJPEG File and drag'n'drop it into ODC. If it plays, MJPEG is supported.

It is more likely that your pipeline is set up incorrectly or that the camera requires special drivers to work under Linux. To rule out ODC as a source of error you could test your pipeline individually using the `gst-launch-1.0` command, for example like this.

```
v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! progressreport update-freq=1 ! \ ! fakesink
```

If that command does not exist immediately the pipeline is working and the camera is running. You should see a progress update by gstreamer every second, but not video as the frames get discarded immediately by the fakesink.

See also 

- https://gstreamer.freedesktop.org/documentation/coreelements/fakesink.html?gi-language=c
- https://gstreamer.freedesktop.org/documentation/debug/progressreport.html?gi-language=c

munsterlander, 2021-11-25:
I have been super busy and hadn't had a chance to come in and update this ticket.  I learned of GST Launch while I was searching and sure enough I was able to run the GStreamer pipeline as written and its throwing errors.  I also tried pipelines that other people had posted for MJPEG but when you run it, ODC just pumps out video-stream stopped so I figure its not getting it into the right format or stream that ODC needs.  I hope to work on it a bit more this weekend though and can post back any errors and issues.

vsaw, 2021-12-08:
I assume you got it working as I have not heard back, if not feel free to re-open.

closed 2021-12-08 by vsaw

munsterlander, 2021-12-09:
I haven't got it working yet... I'm very close. Unfortunately, work has me working on the python portion, so I'm trying to find time to cycle back.  This can be closed.  I don't believe there is an ODC gitter or forum or anything, right?  I hate to just create issues when trying to figure something out.

vsaw, 2021-12-09:
> I haven't got it working yet... I'm very close. Unfortunately, work has me working on the python portion, so I'm trying to find time to cycle back.  This can be closed.  I don't believe there is an [...]

No but we could open the discussions feature in GitHub to filter support questions maybe it's worth a try 🤷‍♂️

munsterlander, 2021-12-09:
Might be worthwhile - but then its more places to monitor for support.  I just feel bad opening issues when its really more of a, has anyone experienced X while doing this and it may not even be an ODC specific thing.  Like for this issue, I know I did it - its not ODC.  Its this specific wireless camera receiver that only provides that one output and my limited use of GStreamer that is the issue.  I just feel bad artificially inflating the issue count for ODC when its not necessary, but I have nowhere else to go to ask these questions - that I am aware of that will have a high percentage of being answered correctly.