opendatacam/opendatacam #501: Problem counting the vehicles (Wrong direction is counted up)

dennbas, 2021-11-26:
Hello together, 

I'm running Opendatacam and Yolov4 on Jetson AGX Xavier with this configuration ["config.json"](https://github.com/opendatacam/opendatacam/files/7608110/config.txt) (.json renamed to .txt).
I have currently this problem:
In this video I have two counting lines that have different direction. In some scenes the car was also counted additional in the wrong driving direction. I could not figured out, what kind of error can this be.

https://user-images.githubusercontent.com/93678091/143570487-fa314b80-1910-4b52-adf3-b22ed01ddbd3.mp4

https://user-images.githubusercontent.com/93678091/143570896-94e4c51d-d413-4e6e-97bc-474953090e16.mp4

vsaw, 2021-11-27:
My guess is, that if you evaluate the tracker data you will find that the center of the bounding box will be moving backwards because of inaccuracies in the detection of the object (notice how the bounding box is jumping even when the car is not moving?)

Currently there is no way to define a "minimum travel distance" or "hysteresis" to prevent this. 

However have you tried to set up one bidirectional counting line instead of two? As it gives you the direction of the vehicle in the JSON/CSV file you should be able to collect the same data but possibly without the errors.

haviduck, 2021-11-30:
> My guess is, that if you evaluate the tracker data you will find that the center of the bounding box will be moving backwards because of inaccuracies in the detection of the object (notice how the bounding box is jumping even when the car is not moving?)
> 
> Currently there is no way to define a "minimum travel distance" or "hysteresis" to prevent this.
> 
> However have you tried to set up one bidirectional counting line, at a location where the vehicles are still moving sufficiently fast, instead of two? As it gives you the direction of the vehicle in the JSON/CSV file you should be able to collect the same data but possibly without the errors.

yeah, ill bet my hat that you right. i faced this when i used the previous coordinate against the current coordinate to check tripline intersection. i store around 20 frames of trajectoryhistory, so my solution was to check first recorded coordinate against the current one instead of using only the previous one. yolo bboxes be jumpy...