opendatacam/opendatacam #502: Support for JetPack 4.6

vsaw, 2021-12-04:
I tried running ODC on JetPack 4.6. The docker version fails to run from USB camera with the following error

```
[ WARN:0] global /usr/src/app/opencv-4.1.1/modules/videoio/src/cap_gstreamer.cpp (1757) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Failed to allocate required memory.
```

I currently don't know how to fix this. 🤷‍♂️

munsterlander, 2021-12-04:
Are you running a usb 3.0 device?

vsaw, 2021-12-04:
No I'm connecting a USB 2.0 camera. The gst pipeline does seem to work outside of Docker. But not within docker for whatever reason.

I briefly got the camera to run in docker, but have no clue how and can not reproduce it. 🤷‍♂️

vsaw, 2021-12-04:
Update this also fails on v3.0.2 with JetPack 4.5.1

munsterlander, 2021-12-05:
That is very very interesting.  I have not had the issue at all running 4.6.  I wonder if its purely a device issue.  Do you have another cam to check?  I have hot swapped cams, started it with USB, etc. and have not experienced this issue.  I am not running the docker build though.

vsaw, 2021-12-05:
I just have the one USB camera, so I can not rule out device issues.

What I can tell is besides unplugging power cycling the USB hub via SW also fixes the camera, at least for JP 4.5.1

As I said this also happened with 4.6 but I have not tested it so thoroughly as I installed 4.5.1 quickly (who I thought was working)

vsaw, 2021-12-08:
I could test with another camera. The other camera worked out of the box.

Interestingly enough, the original camera also works on a Jetson Nano 2 GB (JP 4.6) on the USB 2 Ports without any problems. I will therefore consider this an camera issue and consider this closed.

closed 2021-12-08 by vsaw