opendatacam/opendatacam #503: Update Installation Instructions for Jetson Nano 2 GB

vsaw, 2021-12-08:
Should be a simple update of documentations. Following 2 items are specific to the Jetson nano 2 GB that need to be included in the documentation.

- Emphasize Headless Installation
- Create Swap file

munsterlander, 2022-02-02:
@vsaw This is what I put down in my own docs when I was building out ODC so I could have a repeatable process.  Would you recommend they use the scripts mentioned in these links and issues for swap files?

- Once booted, a few things, do not automatically update software - EVER.

- Lets create a swap file so we have access to more memory:https://www.jetsonhacks.com/2019/04/14/jetson-nano-use-more-memory/
- Be aware of this: https://forums.developer.nvidia.com/t/jetpack-4-2-3-has-a-built-in-2gb-swap-file-we-are-not-able-to-increase-this-to-4-gb/108324/5

vsaw, 2023-11-29:
Close via https://github.com/opendatacam/opendatacam-website/pull/104/files#diff-fd65949b158149b642fdaf61905809394f55d5cbbd37717fb19a57ac6b61dfceR13-R16

closed 2023-11-29 by vsaw