opendatacam/opendatacam #510: YOLO-V5 - Support

SerdarHelli, 2022-01-25:
Does opendatacam support to YOLOV5 ? Is there any way to implement ? If there is , how can I implement ? Is it like others Yolo's mentioned in the documentations ?

SerdarHelli, 2022-01-25:
Actually ,  I was wondering  that  if  I want to change the model intstead of yolov4 or yolov3 , can I  do it in the future ?

vsaw, 2022-01-25:
I hope I understand your question correctly. If you want to use different models/weights (e.g. training your own model) that is already possible. See here: https://github.com/opendatacam/opendatacam/blob/master/documentation/CONFIG.md#use-custom-neural-network-weights

P.S. As this feels more like a support question, instead of discussing development opportunities for ODC I will convert this into a GitHub Discussion. But please keep asking. And if I'm wrong and you really want to make changes to the ODC code please open a new issue.