opendatacam/opendatacam #544: Recording live footage

nishantsah08, 2022-06-04:
Is there a way to store live footage coming from ip camera or final processed video?
If not, what will be the optimal way to do it?

Currently I m using 4gb nano model.

vsaw, 2022-06-04:
That feature does not exist in OpenDataCam. The simplest way would be to use `gst-launch` directly. I haven't tried it but this command should work

```
gst-launch-1.0 <plug in your IP camera here> ! videoconvert ! x264enc tune=zerolatency pass=quant !  h264parse ! mp4mux ! filesink location=video.mp4 -e
```

From https://github.com/jackersson/awesome-gstreamer/blob/master/cheat_sheet/commands.md#from-webcam

nishantsah08, 2022-06-06:
Thanks for the pointer. This is what i ended up using. It rolls the files every 1 min
gst-launch-1.0 rtspsrc location='rtsp://admin:abcd1234@192.168.1.64/Streaming/Channels/101' drop-on-latency=true latency=0 ! rtph264depay ! h264parse ! splitmuxsink location=video%03d.mp4 max-size-time=60000000000

vsaw, 2022-06-10:
Thank you for sharing your solution. I'll close this issue for now. Feel free to reopen if it closed it prematurely.

closed 2022-06-10 by vsaw