opendatacam/opendatacam #559: Fail building on jetson nano

frsisalima, 2022-07-18:
Message error is:
standard_init_linux.go:228: exec user process caused: no such file or directory

Jetson 4gb

Error is only when use FROM opendatacam/opendatacam-darknet-base:v1.0.0-odc-Yolo-v3-nano in Dockerfile
with FROM resinplayground/jetson-nano-cuda-cudnn-opencv:v0.2-slim(last versions)  work ok.

vsaw, 2022-07-18:
I assume you are building the master branch?

The current master does not build because of signing key changes on NVIDIA side, See https://github.com/opendatacam/opendatacam/issues/541. However that change is already fixed in development, so I suggest you try to build from development branch.

If you think this may be something else, please provide more logs. Thank you.

frsisalima, 2022-07-18:
Yes with master branch, now i have tried in the dev branch and work ok with FROM opendatacam/opendatacam-darknet-base:v1.0.1-odc-Yolo-v3-nano

closed 2022-07-18 by frsisalima