opendatacam/opendatacam #560: Yolov7

tdurand, 2022-08-23:
Just in case you don't follow the Yolo world news, Yolov7 seems to improve quite a bit the detection performance / speed , which is the next version after the one we use, yolov4  _(yolov5 and yolov6 being from different author / researcher)_

https://github.com/AlexeyAB/darknet/issues/8595

It **seems** that it would be very low effort as it is compatible with darknet for inference, at least from what I read for Yolov7-tiny .

As we maintain a very light fork from the darknet: https://github.com/opendatacam/darknet/pull/7 , it might be a "quick" win

All that said I don't have time to focus on that myself before end of the year.. but I though it would be nice to share in case someone wants to experiment with it.. I'm if there is a substantial improvement in FPS on the jetson nano 👀

TychoML, 2023-04-21:
I would be much interested in support of YOLOv7 in ODC as there is a definite speed up, excellent for edge devices. It is also easier to train as an added bonus (nice for those into making custom models).

vsaw, 2023-07-04:
@TychoML I'll be taking a look at this. If we really can get YOLOv7 running on the current Darknet implementation, I will include it into the upcoming v4 release. If we need to replace Darknet, it will have to wait.

Have you tried running the YOLOv7 weights in ODC?

vsaw, 2023-08-01:
I'm currently trying to get Yolov7 running on OpenDataCam. Here are some preliminary results:

The current Darknet fork we use can not run yolov7. The upstream version however runs Yolov7 without issues.

As for FPS I got mixed results. v7 seems significantly slower then v4 in the tiny model.

| **Model Size** | **v4 fps** | **v7 fps** | **diff %** |
|--|--|--|--|
| tiny | 60 | 40 | -33 % |
| "regular" | 3.9 | 4.1 | +5 % |

The accuracy, at first glance does not show much of a difference between v4 and v7.

vsaw, 2023-08-01:
Quick follow up benchmark using video from https://motchallenge.net/vis/MOT17-04-SDP shows that the pre-trained model of YOLOv7 performs worse than v4.

YOLOv4

[image]

YOLOv7

[image]

I don't know if this is because of YOLOv7 or because of the way how the weights got retrained from Pytorch to run in Darknet. But given the preliminary results. I don not think a switch to YOLOv7 is justified. I will therefore close this until someone can show me how to get YOLOv7 run properly in Darknet.

closed 2023-08-01 by vsaw