opendatacam/opendatacam #568: OpenDataCam - Xavier NX Ubuntu 20.04 JetPack 5.0.2

effort10, 2022-10-28:
Hi,

I recently wanted to use ubuntu 20.04 on my Xavier NX so i flashed the device with SDK Manager without any issues but when it comes to launching opendatacam i get the below errors on console when i try start opendatacam.

[image]

I have tried googling but cant find out what the issue is.
Any help would be appreciated.

Thank you

vsaw, 2022-11-01:
This looks like OpenDataCam can not start the Darknet Process.

As the Xavier NX is not yet officially supported you would need to compile Darknet manually to run OpenDataCam on it. Did you do this? Or how exactly did you get the Darknet version you are running?

effort10, 2022-11-18:
Thank you

We will have to manually compile darknet as the installer didn't install it on JP5.(think its CUDA version related)
Will let you know once its a success.

Thank you for your assistance.
Regards

vsaw, 2023-02-15:
Closed due to inactivity.