opendatacam/opendatacam #579: Hello Darknet My old Friends

anasBahou, 2023-02-09:
- [ x] avialable now

[image]
@b-g @tdurand "Anas BAHOU"

vsaw, 2023-02-09:
Hi Anas, could you please help me understand what this issue is about? thank you?

vsaw, 2023-02-18:
Im going to close this due to inactivity. Also I saw you posted the exact same message here https://github.com/opendatacam/opendatacam/issues/112#issuecomment-1423577702

closed 2023-02-18 by vsaw

anasBahou, 2023-03-11:
I was just stating that I am free to volunteer on ODC project, especially in on Deepstream as an alternative to Darknet which is available on this old [repo](https://github.com/opendatacam/opendatacam-deepstream).
Let me know if  I can help improve the ODC project.

Thanks in advance.
@b-g @tdurand @vsaw

vsaw, 2023-03-16:
Hi @anasBahou,

thats great to hear! Any help is appreciated! Here are my thoughts on what may be a good way to get started:

Yes, you are right, Darknet is getting out old and there are better alternatives out there by now. However I'm a bit reluctant to simply swap out Darknet for Deepstream for 3 reasons:

- A lot of people have spent much time developing custom Darknet models for their applications, so we should not just abandon them
- People have been asking to run PyTorch and other engines with Darknet, so focusing on DeepStream will not help them run the models and alogrithms they want/need
- Deepstream will limit our ability to bring OpenDataCam to other HW platforms without NVIDIA GPUs, e.g. RaspberryPi, Google Coral or plain old CPUs (for where you don't need real-time capabilities)

Therefore I propose we should aim for a "pluggable Neural Network" architecture where OpenDataCam defines a clean API, that we can use to plug in, Darknet, DeepStream any other Neural Network engine.

As a first step, we would need to clean up the current code a bit as there is currently no clear API and Darknet leaks, all over the place. I've created this issue a while ago to track the work, if you want to get started, please feel free to go ahead https://github.com/opendatacam/opendatacam/issues/342