opendatacam/opendatacam #584: MongoDB persistTracker=false setting not applied by OpenDatacam

effort10, 2023-03-26:
Hi,

It seems that Opendatacam is not applying the persistTracker:false on the development branch installation.
I am still getting records being written to the mongodb tracker collection for some reason.

Has anybody else noticed this issue.

Thanks

[image]

vsaw, 2023-03-26:
I quickly skimmed through the code and it should still work. Are you running development from source or one of the docker images?

And just to make sure the rest of the config works? So I you make changes they get accepted?

effort10, 2023-03-27:
Hi. We ran git clone from git clone --branch development https://github.com/opendatacam/opendatacam.git
and then installed it via install-opendatacam.sh, yes it was a docker install, should i rather do non-docker?

Thank you

vsaw, 2023-03-27:
Ah that explains it. The standard docker-compose.yml files always point to the latest non-development release!

Just edit them and replace the docker tag, this should then pull the correct docker image!

effort10, 2023-03-27:
Hi,

Makes sense, thank you, i will try that.

Regards

closed 2023-03-28 by vsaw