opendatacam/opendatacam #585: Opendatacam output resolution

effort10, 2023-04-10:
Good Day,

is there any config that could allow a different output resolution whilst accessing opendatacam app. Eg http://localhost:8080

This would be for trying to configure a device remotely in an area that has bad 4G/5G signal so drawing the counting lines over jittery video would make it hard to see if everything works 100%. 

Is this something i could just do myself in the code? 

Another extra idea would be to output just the counter lines as a layer on the video feed. 

Similar to the mpeg stream that is exposed on port 8090 but it would just have the counter lines as a layer on top. 

Thanks. :)

vsaw, 2023-04-11:
Hey, I'll try to answer all your questions in line below :-)

> is there any config that could allow a different output resolution whilst accessing opendatacam app. Eg http://localhost:8080

If you mean the resolution of the video feed, then the answer is sadly "no". There's nothing built into ODC to change the video feed resolution.

> This would be for trying to configure a device remotely in an area that has bad 4G/5G signal so drawing the counting lines over jittery video would make it hard to see if everything works 100%.

What exactly do you mean? The video and boxes use two different streams (which may sometimes cause them to be out of sync). For counting, only the boxes and path are relevant, so even if the video is out of sync you should be able to see the trajectories of the objects from the pathfinder view, which should help you place the counter lines.

> Is this something i could just do myself in the code?

Maybe, depending on your coding skills. A starting point would be the mjpegproxy file, which forwards the images from Darknet to the browser.

> Another extra idea would be to output just the counter lines as a layer on the video feed.
> 
> Similar to the mpeg stream that is exposed on port 8090 but it would just have the counter lines as a layer on top.

Not sure I understand. Can you maybe create a mock of that so I can see what you mean?

maddust, 2023-04-11:
hello @effort10 . 

I think i have faced what are you describing... ODC requires a lot of bandwidth to be able to stream raw mjpeg on the browser . Some additional tips adding to @vsaw reply: 

1) lower your source video input resolution: darknet resizes every frame to the resolution the model has been trained on anyways so accuracy should not be affected. Ex. 416 x 416 on the standard coco dataset.  so you should be ok even with a 360p video with a decent bitrate from your video source.  

this should also help with the video stream via 4G/5G connection  in the brower and also increase FPS. 

2)Use the pathfinder trajectories to determine where your lines or zone should be placed. adding to that i recommend to install a Page Marker or Draw on Web extention on your brower.  this will allow you to mark over your trajectories .  so when you switch to the counter view you still have the reference where to place your lines.

3) For counter lines as a layer on the video feed idea: thats totally doable with openCV by downloading your line setup json file and making a script to overlay the line coordinates...  but in that case, you have to already have set the lines on site or on your local network....

effort10, 2023-04-20:
Hi @maddust  & @vsaw 

Apologies for the late reply.
I will try all the options that you guys have provided above :)

Thank you for the great suggestions.