opendatacam/opendatacam #614: Detection in 3.0.2 is not working on desktop

hahahannes, 2023-09-28:
Hi!

Thanks a lot for this cool project! Unfortunately, I am having problems getting it to run.

Expected:
Get detection boxes on the demo video file.

Actual:
[image]

Steps to reproduce:
I followed the instructions to install the desktop version using the latest release of 3.0.2 and then started the containers.
I did not change the default config.

My setup:
GPU: NVIDIA RTX A2000 8GB Laptop GPU
CUDA-version: 10020 (12020), cuDNN: 7.6.5, GPU count: 1  
 OpenCV version: 4.1.1
Firefox 118
Ubuntu 22.04

Anyone had the same problems? There is definitively something wrong with some versions or my hardware, I guess. 

Logs:
```bash
 > OpenDataCam@3.0.0 start /var/local/opendatacam
test-opendatacam-1  | > PORT=8080 NODE_ENV=production node server.js
test-opendatacam-1  | 
test-opendatacam-1  | -----------------------------------
test-opendatacam-1  | -     Opendatacam initialized     -
test-opendatacam-1  | - Config loaded:                  -
test-opendatacam-1  | {
test-opendatacam-1  |   "OPENDATACAM_VERSION": "3.0.0",
test-opendatacam-1  |   "PATH_TO_YOLO_DARKNET": "/var/local/darknet",
test-opendatacam-1  |   "VIDEO_INPUT": "file",
test-opendatacam-1  |   "NEURAL_NETWORK": "yolov4",
test-opendatacam-1  |   "VIDEO_INPUTS_PARAMS": {
test-opendatacam-1  |     "file": "opendatacam_videos/demo.mp4",
test-opendatacam-1  |     "usbcam": "v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink",
test-opendatacam-1  |     "raspberrycam": "nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink",
test-opendatacam-1  |     "remote_cam": "YOUR IP CAM STREAM (can be .m3u8, MJPEG ...), anything supported by opencv",
test-opendatacam-1  |     "remote_hls_gstreamer": "souphttpsrc location=http://YOUR_HLSSTREAM_URL_HERE.m3u8 ! hlsdemux ! decodebin ! videoconvert ! videoscale ! appsink"
test-opendatacam-1  |   },
test-opendatacam-1  |   "TRACKER_SETTINGS": {
test-opendatacam-1  |     "objectMaxAreaInPercentageOfFrame": 80,
test-opendatacam-1  |     "confidence_threshold": 0.2,
test-opendatacam-1  |     "iouLimit": 0.05,
test-opendatacam-1  |     "unMatchedFrameTolerance": 5
test-opendatacam-1  |   },
test-opendatacam-1  |   "COUNTER_SETTINGS": {
test-opendatacam-1  |     "minAngleWithCountingLineThreshold": 5,
test-opendatacam-1  |     "computeTrajectoryBasedOnNbOfPastFrame": 5
test-opendatacam-1  |   },
test-opendatacam-1  |   "VALID_CLASSES": [
test-opendatacam-1  |     "*"
test-opendatacam-1  |   ],
test-opendatacam-1  |   "DISPLAY_CLASSES": [
test-opendatacam-1  |     {
test-opendatacam-1  |       "class": "bicycle",
test-opendatacam-1  |       "hexcode": "1F6B2"
test-opendatacam-1  |     },
test-opendatacam-1  |     {
test-opendatacam-1  |       "class": "person",
test-opendatacam-1  |       "hexcode": "1F6B6"
test-opendatacam-1  |     },
test-opendatacam-1  |     {
test-opendatacam-1  |       "class": "truck",
test-opendatacam-1  |       "hexcode": "1F69B"
test-opendatacam-1  |     },
test-opendatacam-1  |     {
[... 96 lines cut ...]
test-opendatacam-1  |    0 (node:33) [DEP0001] DeprecationWarning: OutgoingMessage.flush is deprecated. Use flushHeaders instead.
MODEL_PARAMETERS.....
test-opendatacam-1  | [yolo] params: iou loss: ciou (4), iou_norm: 0.07, cls_norm: 1.00, scale_x_y: 1.05
test-opendatacam-1  | Total BFLOPS 60.137 
test-opendatacam-1  | avg_outputs = 500162 
test-opendatacam-1  | Demo
test-opendatacam-1  | net.optimized_memory = 0 
test-opendatacam-1  | mini_batch = 1, batch = 8, time_steps = 1, train = 0 
test-opendatacam-1  | nms_kind: greedynms (1), beta = 0.600000 
test-opendatacam-1  | nms_kind: greedynms (1), beta = 0.600000 
test-opendatacam-1  | nms_kind: greedynms (1), beta = 0.600000 
test-opendatacam-1  |  Allocate additional workspace_size = 52.43 MB 
test-opendatacam-1  | Loading weights from yolov4.weights...Done! Loaded 162 layers from weights-file 
test-opendatacam-1  | Got first message from JSONStream
test-opendatacam-1  | Error with message send by YOLO, not valid JSON
test-opendatacam-1  | setvideoresolution
test-opendatacam-1  | 
test-opendatacam-1  |  seen 64, trained: 32032 K-images (500 Kilo-batches_64) 
test-opendatacam-1  | video file: opendatacam_videos/demo.mp4
test-opendatacam-1  | Video stream: 640 x 360 
test-opendatacam-1  | JSON_sender: new client 30
test-opendatacam-1  | MJPG_sender: new client 32
test-opendatacam-1  | Error with message send by YOLO, not valid JSON

```

vsaw, 2023-09-28:
Hi @hahahannes could you please provide a bit more logs?

1. Can you show additional output from Darknet to see if it's starting up correctly?
1. Once OpenDataCam is running in the Browser, could you open port 8070 in your browser to show if Darknet is sending a JSON stream of detections?

And one more question: Do you see any moving boxes at all?

hahahannes, 2023-09-28:
Yes, sure. I am actually trying again with 3.0.2 as I first posted logs with 3.0.0.

A few additional notes:
Loading the model takes a long time. First when I reloaded the page, I got:
```bash
opendatacam-opendatacam-1  | alreadt started
opendatacam-opendatacam-1  | Something went wrong: connect ECONNREFUSED 172.21.5.2:8070
opendatacam-opendatacam-1  | Too much retries, YOLO took more than 3 min to start, likely an error
opendatacam-opendatacam-1  | 0
```
Which is to be expected, I guess, as it is probably still loading.
But if I wait long enough, I get the model parameters logged and:
```bash
opendatacam-opendatacam-1  | [yolo] params: iou loss: ciou (4), iou_norm: 0.07, cls_norm: 1.00, scale_x_y: 1.05
opendatacam-opendatacam-1  | Total BFLOPS 60.137 
opendatacam-opendatacam-1  | avg_outputs = 500162 
opendatacam-opendatacam-1  | Demo
opendatacam-opendatacam-1  | net.optimized_memory = 0 
opendatacam-opendatacam-1  | mini_batch = 1, batch = 8, time_steps = 1, train = 0 
opendatacam-opendatacam-1  | nms_kind: greedynms (1), beta = 0.600000 
opendatacam-opendatacam-1  | nms_kind: greedynms (1), beta = 0.600000 
opendatacam-opendatacam-1  | nms_kind: greedynms (1), beta = 0.600000 
opendatacam-opendatacam-1  |  Allocate additional workspace_size = 52.43 MB 
opendatacam-opendatacam-1  | Loading weights from yolov4.weights...Done! Loaded 162 layers from weights-file 
Got first message from JSONStream
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | setvideoresolution
opendatacam-opendatacam-1  | 
opendatacam-opendatacam-1  |  seen 64, trained: 32032 K-images (500 Kilo-batches_64) 
opendatacam-opendatacam-1  | video file: opendatacam_videos/demo.mp4
opendatacam-opendatacam-1  | Video stream: 640 x 360 
opendatacam-opendatacam-1  | JSON_sender: new client 31
opendatacam-opendatacam-1  | Restore counting areas
opendatacam-opendatacam-1  | MJPG_sender: new client 32
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON

```

1. Are these the logs from above or can I find them in the container?
2. Yes I get lot of output 
```json
[
{
 "frame_id":899, 
 "objects": [ 
  {"class_id":0, "name":"person", "relative_coordinates":{"center_x":1.001923, "center_y":0.155769, "width":5.257868, "height":140491685888.000000}, "confidence":1.000000}, 
  {"class_id":13, "name":"bench", "relative_coordinates":{"center_x":1.001923, "center_y":0.155769, "width":5.257868, "height":140491685888.000000}, "confidence":1.000000}, 
  {"class_id":56, "name":"chair", "relative_coordinates":{"center_x":1.001923, "center_y":0.155769, "width":5.257868, "height":140491685888.000000}, "confidence":0.999511}, 
  {"class_id":57, "name":"sofa", "relative_coordinates":{"center_x":1.001923, "center_y":0.155769, "width":5.257868, "height":140491685888.000000}, "confidence":1.000000}, 
  {"class_id":73, "name":"book", "relative_coordinates":{"center_x":1.001923, "center_y":0.155769, "width":5.257868, "height":140491685888.000000}, "confidence":0.933311}, 
  {"class_id":0, "name":"person", "relative_coordinates":{"center_x":0.922262, "center_y":0.844231, "width":0.000028, "height":0.121144}, "confidence":1.000000}
 ] 
}, 
```
3. The boxes from the screenshot do move, but in a strange way. One is just getting smaller and smaller.

vsaw, 2023-09-28:
@hahahannes Thank you for the logs, those are sufficient, yes!

My assumption is you running into https://github.com/opendatacam/opendatacam/issues/368. This is fixed in development, but not released yet.

If you want, you can try the development build and see if this fixes the issue. If you use the development build, please be aware that the config.json has been updated and needs to be migrated. Take a look at the `config.json` in the development branch to see which changes need to be made to your config. https://github.com/opendatacam/opendatacam/blob/development/config.json

hahahannes, 2023-09-28:
Alright, I will try! Thank you very much for your help!

vsaw, 2023-09-28:
@hahahannes Ah wait a second, I just took a closer look at your JSON output and those don't make any sense e.g. there's no sofa in the scene and the numbers should in the range between 0 and 1 as those are relative to the image size). Maybe it's a Darknet issue after all.

You you try to use `yolov4-tiny` and see what happens? In your config set `NEURAL_NETWORK` to `yolov4-tiny` to try this.

hahahannes, 2023-09-28:
Yes good catch! The problem also occurs with the development image.

hahahannes, 2023-09-28:
Okay, with yolov4-tiny, I am stuck at the loading screen and get:
```bash
[yolo] params: iou loss: ciou (4), iou_norm: 0.07, cls_norm: 1.00, scale_x_y: 1.05
opendatacam-opendatacam-1  |   31 route  27 		                           ->   13 x  13 x 256 
opendatacam-opendatacam-1  |   32 conv    128       1 x 1/ 1     13 x  13 x 256 ->   13 x  13 x 128 0.011 BF
opendatacam-opendatacam-1  |   33 upsample                 2x    13 x  13 x 128 ->   26 x  26 x 128
opendatacam-opendatacam-1  |   34 route  33 23 	                           ->   26 x  26 x 384 
opendatacam-opendatacam-1  |   35 conv    256       3 x 3/ 1     26 x  26 x 384 ->   26 x  26 x 256 1.196 BF
opendatacam-opendatacam-1  |   36 conv    255       1 x 1/ 1     26 x  26 x 256 ->   26 x  26 x 255 0.088 BF
opendatacam-opendatacam-1  |   37 yolo
opendatacam-opendatacam-1  | [yolo] params: iou loss: ciou (4), iou_norm: 0.07, cls_norm: 1.00, scale_x_y: 1.05
opendatacam-opendatacam-1  | Total BFLOPS 6.910 
opendatacam-opendatacam-1  | avg_outputs = 310203 
opendatacam-opendatacam-1  |  Allocate additional workspace_size = 26.22 MB 
opendatacam-opendatacam-1  | Loading weights from yolov4-tiny.weights...Demo
opendatacam-opendatacam-1  | net.optimized_memory = 0 
opendatacam-opendatacam-1  | mini_batch = 1, batch = 1, time_steps = 1, train = 0 
opendatacam-opendatacam-1  | nms_kind: greedynms (1), beta = 0.600000 
opendatacam-opendatacam-1  | nms_kind: greedynms (1), beta = 0.600000 
opendatacam-opendatacam-1  | Done! Loaded 38 layers from weights-file 
```
and when I reload the page
```
already started
opendatacam-opendatacam-1  | Got first message from JSONStream
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | JSON_sender: new client 32
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
opendatacam-opendatacam-1  | Error with message send by YOLO, not valid JSON
```

hahahannes, 2023-09-28:
But I get predictions
```json
  {"class_id":8, "name":"boat", "relative_coordinates":{"center_x":0.191346, "center_y":0.922115, "width":3.482846, "height":0.000380}, "confidence":1.000000}, 
  {"class_id":32, "name":"sports ball", "relative_coordinates":{"center_x":0.191346, "center_y":0.922115, "width":3.482846, "height":0.000380}, "confidence":1.000000}, 
  {"class_id":56, "name":"chair", "relative_coordinates":{"center_x":0.191346, "center_y":0.922115, "width":3.482846, "height":0.000380}, "confidence":1.000000}, 
  {"class_id":59, "name":"bed", "relative_coordinates":{"center_x":0.191346, "center_y":0.922115, "width":3.482846, "height":0.000380}, "confidence":1.000000}, 
  {"class_id":2, "name":"car", "relative_coordinates":{"center_x":0.152885, "center_y":0.922115, "width":2.033396, "height":0.000995}, "confidence":1.000000}, 
  {"class_id":6, "name":"train", "relative_coordinates":{"center_x":0.152885, "center_y":0.922115, "width":2.033396, "height":0.000995}, "confidence":1.000000}, 
  {"class_id":8, "name":"boat", "relative_coordinates":{"center_x":0.152885, "center_y":0.922115, "width":2.033396, "height":0.000995}, "confidence":1.000000}, 
  {"class_id":32, "name":"sports ball", "relative_coordinates":{"center_x":0.152885, "center_y":0.922115, "width":2.033396, "height":0.000995}, "confidence":1.000000}, 
  {"class_id":56, "name":"chair", "relative_coordinates":{"center_x":0.152885, "center_y":0.922115, "width":2.033396, "height":0.000995}, 
```

vsaw, 2023-09-28:
Hmm, the predictions are still not working correctly. I don't quite get why Darknet is spitting out confused predictions. I don't have a quick and ready solution for you at the moment. Debugging and re-building the Desktop image is going to take time. Sorry for that.

As a workaround I suggest you try running OpenDataCam and Darknet without Docker, if that's an option for you.

hahahannes, 2023-09-28:
No worries! Thanks for your help so far! 

Will try without docker, maybe there is something going wrong with the runtime. Could also be my cuda setup. Lets see!

cguerra, 2023-10-03:
Hola, lo pudiste solucionar ?, me pasa exactamente lo mismo con una nvidia RTX 3070

vsaw, 2023-10-04:
@cguerra I'm not aware of any solution right now. I will try to take a look at this this week and squeeze in a fix for the development branch.

vsaw, 2023-10-10:
I just quickly tried the development docker image on an AWS g4dn.xlarge running the [AWS Deep Learning Base GPU AMI (Ubuntu 20.04)](https://aws.amazon.com/releasenotes/aws-deep-learning-base-gpu-ami-ubuntu-20-04/) and it all was working out of the box. Biggest difference between your machine and the AWS server I can see is the default CUDA version (AWS uses CUDA 11.8) Maybe CUDA 10.2 (which was released in 2019) is simply to old for ODC to run properly.

And by the way you don't need an account to install cuDNN, there are official Ubuntu repositories available. See https://docs.nvidia.com/deeplearning/cudnn/install-guide/index.html#package-manager-ubuntu-install

vsaw, 2023-10-10:
I consider this issue solved for now and created a follow up issue to track the minimum CUDA requirements in the docs. See https://github.com/opendatacam/opendatacam/issues/618

closed 2023-10-10 by vsaw