opendatacam/opendatacam #635: (node:40) UnhandledPromiseRejectionWarning: TypeError: Cannot set property 'filename' of undefined

sukesh-ak, 2024-04-20:
Hi guys,

Need help. I am trying to get `opendatacam` working for testing on `Ubuntu` running in `WSL2`.
Installation and running docker works fine. But the initial loading phase does not complete with the following exception.

I have manually verified the uploaded file is in the expected folder inside the container.

### Exception details here
```text
(node:40) UnhandledPromiseRejectionWarning: TypeError: Cannot set property 'filename' of undefined
    at /var/local/opendatacam/server.js:990:46
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
(node:40) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:40) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

### docker-compose.yml file here
```yaml
version: "2.3"
services:
  opendatacam:
    restart: always
    runtime: nvidia
    image: opendatacam/opendatacam:v3.0.2-desktop
    privileged: true
    volumes:
      - './config.json:/var/local/opendatacam/config.json'
      - './opendatacam_videos:/var/local/darknet/opendatacam_videos'
    ports:
      - "8080:8080"
      - "8070:8070"
      - "8090:8090"
  mongo:
    image: mongo
    restart: always
    ports:
      - "27017:27017"
    volumes:
      - mongodb_data_container:/data/db
volumes:
  mongodb_data_container:
```

### Complete log below
```log
> Ready on http://localhost:8080
> Ready on http://172.31.0.3:8080
Success init db
(node:40) [DEP0001] DeprecationWarning: OutgoingMessage.flush is deprecated. Use flushHeaders instead.
Process YOLO started
 CUDA-version: 10020 (12040), cuDNN: 7.6.5, GPU count: 1  
 OpenCV version: 4.1.1
 0 : compute_capability = 890, cudnn_half = 0, GPU: NVIDIA GeForce RTX 4060 Laptop GPU 
   layer   filters  size/strd(dil)      input                output
   0 First request on console stream
uploadMulter callback
File upload done
Stop YOLO
Stop recording
Process YOLO exit
Process YOLO stopped
YOLO stopped
/var/local/darknet/opendatacam_videos_uploaded/demo.mp4
Process YOLO initialized
Process YOLO started
(node:40) UnhandledPromiseRejectionWarning: TypeError: Cannot set property 'filename' of undefined
    at /var/local/opendatacam/server.js:990:46
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
(node:40) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:40) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
 CUDA-version: 10020 (12040), cuDNN: 7.6.5, GPU count: 1  
 OpenCV version: 4.1.1
 0 : compute_capability = 890, cudnn_half = 0, GPU: NVIDIA GeForce RTX 4060 Laptop GPU 
   layer   filters  size/strd(dil)      input                output
   0 New client, close previous stream
New client, close previous stream
```

Thanks in advance.

vsaw, 2024-04-20:
This sounds like an error in your `config.json`. Without seeing the file I cannot be more specific than telling you to check how you configured the video input. See also https://opendata.cam/docs/configuration/#video-input

sukesh-ak, 2024-04-20:
> This sounds like an error in your `config.json`. Without seeing the file I cannot be more specific than telling you to check how you configured the video input. See also https://opendata.cam/docs/configuration/#video-input

Thanks for the response.
I am using the default `config.json` and no changes. I see the file `demo.mp4` file in there.

```json
{
  "OPENDATACAM_VERSION": "3.0.2",
  "PATH_TO_YOLO_DARKNET" : "/var/local/darknet",
  "VIDEO_INPUT": "file",
  "NEURAL_NETWORK": "yolov4",
  "VIDEO_INPUTS_PARAMS": {
    "file": "opendatacam_videos/demo.mp4",
    "usbcam": "v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink",
    "raspberrycam": "nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink",
    "remote_cam": "YOUR IP CAM STREAM (can be .m3u8, MJPEG ...), anything supported by opencv",
    "remote_hls_gstreamer": "souphttpsrc location=http://YOUR_HLSSTREAM_URL_HERE.m3u8 ! hlsdemux ! decodebin ! videoconvert ! videoscale ! appsink"
  },
  "TRACKER_SETTINGS": {
    "objectMaxAreaInPercentageOfFrame": 80,
    "confidence_threshold": 0.2,
    "iouLimit": 0.05,
    "unMatchedFrameTolerance": 5
  },
  "COUNTER_SETTINGS": {
    "minAngleWithCountingLineThreshold": 5,
    "computeTrajectoryBasedOnNbOfPastFrame": 5
  },
  "VALID_CLASSES": ["*"],
  "DISPLAY_CLASSES": [
    { "class": "bicycle", "hexcode": "1F6B2"},
    { "class": "person", "hexcode": "1F6B6"},
    { "class": "truck", "hexcode": "1F69B"},
    { "class": "motorbike", "hexcode": "1F6F5"},
    { "class": "car", "hexcode": "1F697"},
    { "class": "bus", "hexcode": "1F68C"}
  ],
  "PATHFINDER_COLORS": [
    "#1f77b4",
    "#ff7f0e",
    "#2ca02c",
    "#d62728",
    "#9467bd",
    "#8c564b",
    "#e377c2",
    "#7f7f7f",
    "#bcbd22",
    "#17becf"
  ],
  "COUNTER_COLORS": {
    "yellow": "#FFE700",
    "turquoise": "#A3FFF4",
    "green": "#a0f17f",
    "purple": "#d070f0",
    "red": "#AB4435"
  },
  "NEURAL_NETWORK_PARAMS": {
    "yolov4": {
      "data": "cfg/coco.data",
      "cfg": "cfg/yolov4-416x416.cfg",
      "weights": "yolov4.weights"
    },
    "yolov4-tiny": {
      "data": "cfg/coco.data",
      "cfg": "cfg/yolov4-tiny.cfg",
      "weights": "yolov4-tiny.weights"
    }
  },
  "TRACKER_ACCURACY_DISPLAY": {
    "nbFrameBuffer": 300,
    "settings": {
      "radius": 3.1,
      "blur": 6.2,
      "step": 0.1,
      "gradient": {
        "0.4":"orange",
        "1":"red"
      },
      "canvasResolutionFactor": 0.1
    }
  },
  "MONGODB_URL": "mongodb://mongo:27017",
  "PORTS": {
    "app": 8080,
    "darknet_json_stream": 8070,
    "darknet_mjpeg_stream": 8090
  }
}
```

sukesh-ak, 2024-04-20:
After multiple tries now it shows its loading the default `demo.mp4` but then I am hitting the issue mentioned here https://github.com/opendatacam/opendatacam/issues/614

`http://localhost:8070` is showing the following so no inference

[image]

```json
{
 "frame_id":36, 
 "objects": [ 
  {"class_id":0, "name":"person", "relative_coordinates":{"center_x":0.968587, "center_y":0.155538, "width":0.142953, "height":1.267735}, "confidence":0.865290}, 
  {"class_id":57, "name":"sofa", "relative_coordinates":{"center_x":0.968587, "center_y":0.155538, "width":0.142953, "height":1.267735}, "confidence":0.838910}, 
  {"class_id":0, "name":"person", "relative_coordinates":{"center_x":0.961556, "center_y":0.861943, "width":0.114604, "height":0.354715}, "confidence":0.911529}, 
  {"class_id":0, "name":"person", "relative_coordinates":{"center_x":0.982083, "center_y":0.207010, "width":0.107570, "height":0.453541}, "confidence":0.703170}
 ] 
}, 
```

Keep in mind, I am using default `config.json` and no changes.

Somewhere something is pointing to the wrong config file or the settings in `config.json` is not getting loaded/assigned.

closed 2024-05-29 by sukesh-ak