opendatacam/opendatacam #636: Pre-Requesits into documentation

filippoadessi, 2024-10-01:
Hi, I'm playing with the installation of this incredible tool but I fail, i think, due to some hardware limitation.
Reading docs I realize that I can install docker image on every machine equipped with every gpu. CUDA and Nvidia GPU seems to be options and not strictly needed to run OpenDataCam.

After running the installar I have only the mongo docker image running and the opendatacam ruines due to lack of nvidia hardware. 
[image]

Should you better define minimum hardware requirements to run OpenDataCam? I want to try it in sport environment in order to catch trajectories of balls and athletes during matches. Is there any chance to install it in a virtual server?

vsaw, 2024-10-05:
With Mongo its most likely a ARam

filippoadessi, 2024-10-06:
My “server” is a VPS system hosted by IONOS. It’s a fully virtualizes machine, i think, running on top of VMWare. I can have a root access to an other machine which is not virtualized but lacks of nvidia hardware.